supermerill/SuperSlicer #1755: UNLOCKED LOCK and WHITE BULLET icons are missing (macos)

liftbag, 2021-10-31:
**Describe the bug**
Starting from the Nightly 417 and up to the most recent, the lock/unlock and white dot icons next to the parameters seem to be disappeared.

**>> Config bundle <<**

[SuperSlicer_config_bundle.ini.zip](https://github.com/supermerill/SuperSlicer/files/7449510/SuperSlicer_config_bundle.ini.zip)

**Expected behavior**

[image]

**Actual behavior**

[image]

**Desktop:**
 - OS: MacOS 12.0.1 Monterey
 - From nigthly macos 417 to 420.

tome9111991, 2021-10-31:
Hello.
You can look at https://github.com/supermerill/SuperSlicer/issues/1729
Looks is a new feature

supermerill, 2021-10-31:
It shouldn't be activated by default...

liftbag, 2021-11-01:
Oops, sorry, I was not aware that there was a specific setting in the latest versions.

[image]

liftbag, 2021-11-01:
I noticed another problem on Macos Monterey.
Starting from the Nightly 416 the tooltips that explain the operation of the White Dots and the padlocks work properly, even all those in the preferences window work. Almost all the tooltips of the configuration parameters instead have disappeared, except for some sporadic case.
They work properly from 415 and previous versions.
If it were a real problem, I will open a new issue if necessary.

supermerill, 2021-11-01:
> I noticed another problem on Macos Monterey. Starting from the Nightly 416 the tooltips that explain the operation of the White Dots and the padlocks work properly, even all those in the preferences window work. Almost all the tooltips of the configuration parameters instead have disappeared, except for some sporadic case. They work properly from 415 and previous versions. If it were a real problem, I will open a new issue if necessary.
> 
>  Registrazione.schermo.2021-11-01.alle.10.06.01.mp4

that's a try for mitigation of #1720 (tooltip hiding too soon)
It's the setting 'use custom tooltip' just below 'display setting'. **Should I uncheck it by default?** (it's currently checked by default on macos but I'm undecided)
The os tooltip is available on the title and the 'custom' one is available when you let your cursor on a not-focus field (but the 'int' one, you need to hover one of the arrows)

liftbag, 2021-11-01:
> that's a try for mitigation of #1720 (tooltip hiding too soon) It's the setting 'use custom tooltip' just below 'display setting'. **Should I uncheck it by default?** (it's currently checked by default on macos but I'm undecided) The os tooltip is available on the title and the 'custom' one is available when you let your cursor on a not-focus field (but the 'int' one, you need to hover one of the arrows)
> 
> mmmm. after looking your video, it seems that it's completely broken on macos...

Yes, seems that the "Use custom tooltips" feature is broken on macos starting from version 416. Deactivating it I don't see any problems.

liftbag, 2021-11-02:
I tried the nightly 422 and the custom tooltips don't seem to be improved on macos.

supermerill, 2021-11-02:
erf.
I revert the change where they won't open on focus. Seems like this isn't the culprit...
I think I'll really have to find a way to install/find a macos to be able to test things.

liftbag, 2021-11-02:
> erf. I revert the change where they won't open on focus. Seems like this isn't the culprit... I think I'll really have to find a way to install/find a macos to be able to test things.
Hackintosh could be fine?

supermerill, 2021-11-02:
> Hackintosh could be fine?

If I have to buy hardware, I'll go directly to a mac mini, to be able to compile&run on an m1. I'll have to do it, sooner or later.

I think I saw a way to have a one-button macos-in-a-vm on linux.
but I don't think it can be launch from a linux VM, have to double-boot.

closed 2021-11-19 by supermerill