supermerill/SuperSlicer #1766: Walls on overhangs not connecting

ristein, 2021-11-01:
**Describe the bug**
If I have an overhanging stairstep (horizontal) that is approx as far as my perimeter count x line width or little bigger, there will be a gap between overhanging wall and wall underneath.
Also there won't be bottom layers (solid infill) but infill instead.

**To Reproduce**
Steps to reproduce the behavior:
1. Open my project
2. Look at layers 9 - 11

**>> Project File <<**
[BugOverhang.zip](https://github.com/supermerill/SuperSlicer/files/7454492/BugOverhang.zip)

**Expected behavior**
Overhanging wall and wall underneath should be connected with solid infill

**Screenshots**
I have three same parts with 2mm horizontal overhang. Perimeter count set to 3, 4 and 5.
With 3 perimeters it works as expected (left).
With 4 perimeters walls are not connected: there should be solid infill as wide as wall thickness and as high as bottom solid layers as when overhang retraces back (at layer 21), but here is also a gap that does not get filled. Also in layer 9 the infill doesn't even touch the wall.
With 5 perimeters the walls touch each other a tiny little bit and infill touches the wall but should be solid as well. In layer 10 the infill is suddenly solid but not so in layer 11.
layer 9: 
[image]
layer 10:
[image]
layer 11:
[image]
layer 21:
[image]
part measures:
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.5735

ristein, 2021-11-02:
thank you for this fast response and hopefully fixing it @supermerill .
[image]
Will this bug also be fixed or is this a different problem?
Here the bottom solid layers are set to 10, but layer 3 to 5 already get not solid infill.
[DeckelmSpart.zip](https://github.com/supermerill/SuperSlicer/files/7459279/DeckelmSpart.zip)

supermerill, 2021-11-02:
That's a different problem with "no solid over X perimeters"
As the first layer has bigger perimeters, it considers these area as over perimeters and so prune the infill after 2 layers.
I think i'll have to not consider the X first layers, this will be safer.

ristein, 2021-11-08:
updated to new version, thank you.
another sample, of strange behavior (yes, I forgot the support, but that's not the point):
[Deckel.zip](https://github.com/supermerill/SuperSlicer/files/7498555/Deckel.zip)
strange cut in infill (might have to do with the parts border is in the first layer):
[image]
bridging partly over solid infill, is this intentional?:
[image]
should there be a second bridging layer? Would be nice if there wouldn't be that cut:
[image]

supermerill, 2021-11-08:
isn't it the over-bridge flow?
Currently, it has to split the surface to apply the modified flow. May change that in the future.

ristein, 2021-11-08:
I didn't change flow nor speed if that's the question

supermerill, 2021-11-08:
[1766_Deckel_100_above_bridge.zip](https://github.com/supermerill/SuperSlicer/files/7499500/1766_Deckel_100_above_bridge.zip)

ristein, 2021-11-09:
yes, sorry. Flow above bridges was 120% in my "creality" preset. 
But the first two screenshots won't get solved by this. Any comment on those?

closed 2021-11-19 by supermerill