supermerill/SuperSlicer #1768: 2076 Day Print (error)

inspectionsbybob, 2021-11-01:
### Version
2.3.56

### Operating system type + version
Win10 Current patches

### Behavior
I Sliced this in Cura and it says 9 Hours...
I slice the SAME file in SS and it comes back with 2000+ Hours....
I have gone through my profile several times trying to find an error but can't.
The object has 3-parts, could that be it?
It comes back with an error "A Toolpath outside the pint area was detected" but I can't find any stray objects or filaments, it fits well within the print area, at least what I can see.

Can someone look at this file and tell me if it is the file or my profiles that are corrupt....

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it her
[3MF_DragonWineGlass.zip](https://github.com/supermerill/SuperSlicer/files/7455719/3MF_DragonWineGlass.zip)
e_

supermerill, 2021-11-01:
Should be fixed in the 2.3.57 branch
[image]

supermerill, 2021-11-01:
Can't reproduce
[image]

supermerill, 2021-11-01:
I have a problem, i can't load your settings from your 3mf.
Please save it from superslicer.

inspectionsbybob, 2021-11-01:
So were you able to open the 3MF file in the Zip?

What is the root cause if you think it will be fixed in 2.3.57??  Is there
a work around (other than using Cura?)

Thanks....

PS... I am really liking SS...  So Far it seems to be picking up the best
of Cura and Prusa.... and is much cleaner UI....

Bob Sisson

supermerill, 2021-11-01:
> So were you able to open the 3MF file in the Zip?

I can open it but there isn't the configuration inside. So it's not useful. I need the project file as saved in superslicer to be able to replicate.

inspectionsbybob, 2021-11-02:
Ok, just tried another version of the same print, (the top half) and got the same weird response.  
This time I save it from SS as a 3MF so here it is...
[glass only.zip](https://github.com/supermerill/SuperSlicer/files/7462022/glass.only.zip)

Cura reads this fine, so not sure where the error is, I am sure it is my fault, but I am having a hard time finding the setting that is whacky...

supermerill, 2021-11-02:
You wrote `G91` but forgot to get back to `G90` before letting the slicer print. So it can't read correctly the gcode, and so can't estimate the duration.

btw, it's also better to include G21 and M82 in your start gcode.

inspectionsbybob, 2021-11-02:
Ok, I will ask for some help here, as my Start G-code is a work in
Progress...

What I am TRYING to do is:

   1. Warm up the BED FIRST as it takes the longest to stabilize
   2. Warm up the Extruder (gives the bed more time to stabilize)
   3. Load the Bed-Profile in S1 (M420 S1)
   4. Set to T0 (I will duplicate into T1 when Both Extruders are active)
   5. Draw a line 10mm in from the edge from Y10 to Y260
   6. Move over a touch
   7. Draw a line back to Y10
   8. Retract so No Blob
   9. Reset Extruder
   10. PRINT !

So My Start G-code currently is:
(This is in Printer Settings, Custom G-Code, Start G-code, there is nothing
in Filament overrides)

G28                                                             ;Home
everyone   (There is a bug in Marlin and IDEX in G28 that can cause
problems with heads bumping each other)
M420 S1                                                      ;Use ABL
stored in Position 1
;
M190 S65                                                   ; set bed temp
and wait for temp to be reached
M109 S215 T0                                            ; set nozzle temp
and wait for temp to be reached
;
T0                                                           ; all for E0
left side
 G90                                                       ; Absolute
positioning
G1 Z2.0 F3000                                        ;Move Z Axis up
G1 X5 Y10 Z0.28                                     ;Move to start position
G1 X5 Y300.0 Z0.28 F1500.0 E15           ;Draw the first line
G1 X5.4 Y300.0 Z0.28 F5000.0               ;Move to side a little
G1 X5.4 Y10 Z0.28 F1500.0 E30             ;Draw the second line
G91                                                         ; relative
motion
G1 E-1 F800                                           ;retract 1
T0 G92 E0                                             ; reset extruder0
;
;T1 stuff will go here

And thats it...

So what would your edits be?

PS.... I don't show a M82 in my G-code reference...

neophyl, 2021-11-02:
M82 is extruder to relative extrusion (which is what you have set under Printer Settings>General)  https://marlinfw.org/docs/gcode/M082.html

inspectionsbybob, 2021-11-02:
The only thing set in Printer settings, General, Advanced is "enable
variable layer height"

"Use Relative E distances"  is -NOT- Set