supermerill/SuperSlicer #1769: Some of the Calibration routines don't put anything on the Platter

inspectionsbybob, 2021-11-01:
### Version
2.3.56

### Operating system type + version
Win10 current patches

### Behavior
Went to setup/Calibrate a new Filament Vendor and thought I would use the built in Calibration routines/Objects. The following routines don't have a way to put something on the Platter.  The Filament TEMPERATURE was the one I wanted, I pulled one from Teaching Tech...

- Bed Level Calibration
- Filament Temperature Calibration

The Calibration Cube is "Unmarked", the good ones have X,Y and Z on the appropriate sides.

The BEST Calibration routines are from TeachingTech: https://teachingtechyt.github.io/calibration.html

_Is this a new feature request?_
NO

supermerill, 2021-11-01:
There is a text box with instructions and a "generate" button at the bottom to generate the geometry & settings.
I don't really understand what is the problem.

inspectionsbybob, 2021-11-02:
Several reboots later I found it.

By DEFAULT, the window opens with the bottom several lines below the edge
of the screen.  I was having memory issues and it wouldn't let me move the
BOX so I couldn't see the generate buttons

Close it as Brain Flatulance

closed 2022-10-03 by inspectionsbybob