supermerill/SuperSlicer #1770: Printer Settings loosing "G-Code Flavor"

inspectionsbybob, 2021-11-02:
### Version
2.3.56

### Operating system type + version
Win10 w/patches

### Behavior
On Saving edits to Printer Settings the "G-Code Flavor" is being corrupted.  Had several instances where the Gcode file won't run, pull it into a Gcode viewer and it says "file does not contain valid Gcode"  In going through the settings (again) I find the the Gcode flavor has been set to something weird away from Marlin.  Has happened several times, not sure the exact sequence that corrupts it.

_Is this a new feature request?_
NO
#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2021-11-02:
may be hard to fix.
what is the problem with the gcode?

inspectionsbybob, 2021-11-02:
The symptom was it would warm up and.... nothing....  Though it was a bad
SD/USB device. took a while to find.  After making some edits to the
profile, make the Gcode file and try to run it and nothing after the warm
up... look through the settings and find the flavor on something weird like
repetier... reset that, reslice and everything works as it should...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

supermerill, 2021-11-02:
Next time you reproduce it, tell me how.