supermerill/SuperSlicer #1771: Ignore tool change for selected body

Pavulon87, 2021-11-02:
**Is your feature request related to a problem? Please describe.**
I have a multi-part 3mf file that should be printed in different colors. It consists of a main drawing which does not change and a second part (e.g. a name) which changes. I use a manual filament change so I would like to keep the number of changes to a minimum. That's why I have to waste time creating a new multi-color model in fusion360 every time. It would be good if you could set the option to ignore color/tool changes for the selected model.

**Describe the solution you'd like**
I would like the selected model to be printed similar to the supports - with the same filament as the element in front of it.

[image]

**Describe how it would work**
Just ignore the wipe tower and tool change for the selected body and use the previous one.

**Describe alternatives you've considered**
Changing the color at the selected height does not work as I need it.

**Additional context**
[image]

For example:
This cylinder should be black, red, blue and yellow without need to cut it in parts.
[image]

supermerill, 2021-11-02:
What you can do:
add another extruder, like a 5th
assign this tower to the 5th
in your toolchange gcode:
`{if next_extruder != 4}T[next_extruder]{endif}`

Does this work?

Pavulon87, 2021-11-02:
[image]
It looks good at first.
But the number of tool changes doesn't look good at all.

supermerill, 2021-11-02:
I have another idea:

Allow the wipetower (imo, it's a good idea to have one even for manual toolchange as it will allow a good restart and less error due to manual purging).
set your object as "wipe object" (right-clic, 'wipe options')
and everything should be good.

Pavulon87, 2021-11-02:
Brilliant idea. It looks to be working fine. I used wipetower but didn't think about it.
[image]

Pavulon87, 2021-11-02:
One note. Unfortunately, this does not work when the ignored element is added as "add part.
[image]
So it is not a perfect solution, but it may be useful already.

Pavulon87, 2021-11-02:
It's even more complicated than I thought at first. Part with another extruder cause print separation. So it's probably a bad idea and I need to get back to drawing it in fusion360
[image]
[image]

Pavulon87, 2021-11-04:
A little update: the idea is possible to make :)
I hard-coded that extruder no. 8 should be replaced with the extruder with the highest (the lowest also would work) number on the current layer.
I changed files libslic3r/Fill/Fill.cpp and libslic3r/Layer.cpp and I'm sure it could be done better, but I have no other ideas.
[image]

[image]
)

[image]

supermerill, 2021-11-04:
But does this work because you have only one change and it's between layers?

Pavulon87, 2021-11-04:
It works quite well, but it certainly has its limitations.

[image]
[image]
[image]
[image]
[image]
[image]