supermerill/SuperSlicer #1773: Z-hop only on top does not activate on second ironing pass

TheBoojah, 2021-11-02:
When using Z-hop only on top, during the second pass of ironing, it does not Z-hop:
[image]

[Reindeer ironing torture test.zip](https://github.com/supermerill/SuperSlicer/files/7463677/Reindeer.ironing.torture.test.zip)

Version 2.3.57.3

Probably the same thing as #1285

supermerill, 2021-11-03:
Oh right.
i moved the ironing from "solid +top" to "top +ironing" when rpusa aaded a "ironing" role, to be consistent with the ironing post-process. But of course, I now have to add ironing as a "top".
I wonder if I keep the first pass as a top fill or revert to solid internal... if it's only for avoiding the z-hop, i don't think it's worth the trouble.

TheBoojah, 2021-11-03:
If the first pass is changed to solid internal infill, will that apply regular speeds? If so it will print a lot faster for me, and perhaps the second pass will be unable to really smooth it out properly.

supermerill, 2021-11-03:
yes. Understood.

closed 2021-11-19 by supermerill