supermerill/SuperSlicer #1775: Initial dance in the start of the gcode

newphreak, 2021-11-02:
**Describe the bug**

Expected behaviour: goes through start script and starts printing directly.

Actual behaviour: After heating up it goes down to the center of the part, moves up, moves to front left corner, down, up then starts printing.

`M109 S205 ; set temperature and wait for it to be reached
M190 S60 ; set bed temperature and wait for it to be reached
**;LAYER_CHANGE
;Z:0.24
;HEIGHT:0.24
;BEFORE_LAYER_CHANGE
;0.24

G1 Z0.24 F18000
;AFTER_LAYER_CHANGE
;0.24
G1 E-0.3 F1800
G1 Z1.02 F18000
M204 S1500
G1 X22.092 Y22.097
M204 S2000
G1 X44.184 Y44.194
G1 Z0.24
G1 E0.3 F1800**
;TYPE:Skirt
;WIDTH:0.44
G1 F2400
G1 X47.204 Y41.175 E0.16223`

**To Reproduce**
Happens to everything i slice, so hopefully my project has some clues if this is not normal behaviour.

**>> Project File <<**
[Voron_Design_Cube_v7_56m_20211102-211326.zip](https://github.com/supermerill/SuperSlicer/files/7463757/Voron_Design_Cube_v7_56m_20211102-211326.zip)

**Desktop (please complete the following information):**
 - OS: Windows
 - Version all the latest, even nightly

**Additional context**
You can find me on discord also: Jon#4961

newphreak, 2021-11-02:
On second though, this may seem odd on my printers because they home on y and x max position instead of min. I think this is where the difference stems from.

supermerill, 2021-11-03:
it's exactly that.
It doesn't know where it is.
so it move at printing height ( mayyybe could be avoided).
It detect a big move, so it retract and move to a safe height (because first layer i think)
then it moves to the printing point.
Here is a problem: i recently split move in two to distinguish accel and decel, but here it's problematic as I don't know the starting point, assuming 0,0.

newphreak, 2021-11-03:
Well, during start the plate should be entirely clear, so it should be quite safe to assume it's at a safe height, and that moving directly to print start position should be safe? I fail to see how the aimless jumping to different places would differ. 
So hopefully that can be more streamlined.

Start script is done before all this, so start position would either be skirt/brim or perimeters of the part? I'm not a coder so this is probably alot more complicated then i can imagine.

supermerill, 2021-11-03:
> so it should be quite safe to assume it's at a safe height

I have multiple people asking for that because their bed isn't safe.

What i have with prusslicer 2.3.1:
````
G1 Z0.200 F10800.000
;AFTER_LAYER_CHANGE
;0.2
G1 E-0.80000 F2100.000
G1 Z0.600 F10800.000
G1 X109.943 Y91.476
G1 Z0.200
```

> I'm not a coder so this is probably alot more complicated then i can imagine.

the problem is less a code issue than everyone having a different expectation, the only good one.

I'll try to improve a bit to avoid "dance".

neophyl, 2021-11-03:
If PS beta doesn't do this then wont the issue go away in Susi once its merged ?

supermerill, 2021-11-03:
I don't think so, as I added more things. I'll see when I merge.

newphreak, 2021-11-04:
Just to make it clear, it's not just the "z-dance", what i wanted solved was the go visit the front left corner then jump back to print was my main oddity here. it has an odd jump "halfway" towards x0 y0, then goes back towards where it's supposed to.

supermerill, 2021-11-04:
should be fixed.
just a note: to prevent the z-dance, you must have a clean void "after layer change" custom gcode. Or it will go to the layer height anyway before calling the macro.

newphreak, 2021-11-07:
It's fixed. Thanks.

closed 2021-11-07 by newphreak