supermerill/SuperSlicer #1778: superslicer gcode-viewer path setting

knickohr, 2021-11-03:
Hi,

where is the path stored in superslicer to launch the gcode-viewer from the superslicer menu list

Got an error :
Internal error: bootst::filesystem::create_directory:  The system cannot find the path: "AEf:\SuperSlicer\superslicer-gcodeviewer.exe

Launching the gcodeviewer from explorer works fine.

supermerill, 2021-11-03:
it's `[path of the superslicer exe directory]/uperslicer-gcodeviewer.exe`

knickohr, 2021-11-03:
Yes, but where is it stored ?

???.ini

supermerill, 2021-11-03:
It's not stored, we ask where we are and add "superslicer-gcodeviewer.exe", in a hardcoded way.

knickohr, 2021-11-04:
So, the question, btw. the answer is incorrect. The driveletter is missing or an unreadable letter.

Launching the gcodeviewer standalone works.

Win 7 x64

supermerill, 2021-11-04:
it says "AEf" for the drive letter?
what is the correct one?

knickohr, 2021-11-04:
In this case, yes. But it changes, sometimes its "" nothing :-(

It should be E:\

supermerill, 2021-11-04:
nothing like ":\SuperSlicer\superslicer-gcodeviewer.exe" ?

knickohr, 2021-11-04:
correct

supermerill, 2021-11-04:
Can you download this debug build
https://github.com/supermerill/SuperSlicer/suites/4257603223/artifacts/110598346
and launch it via the command :
`superslicer_console.exe --loglevel=5`
and make it bug.
then post here the content of the console.
Thanks.

knickohr, 2021-11-04:
The zip is not readable or incorrect

knickohr, 2021-11-04:
I tried it with the installed version 2.3.57

[2021-11-04 18:26:51.801227] [0x00001aa0] [info]    Syncing configuration cache
[2021-11-04 18:26:51.811228] [0x00001aa0] [warning] No such vendor: Anycubic
[2021-11-04 18:26:51.811228] [0x00001aa0] [warning] No such vendor: Artillery
[2021-11-04 18:26:51.811228] [0x00001aa0] [warning] No such vendor: BIBO
[2021-11-04 18:26:51.811228] [0x00001aa0] [warning] No such vendor: Creality
[2021-11-04 18:26:51.821228] [0x00001aa0] [warning] No such vendor: FLSun
[2021-11-04 18:26:51.821228] [0x00001aa0] [warning] No such vendor: INAT
[2021-11-04 18:26:51.831228] [0x00001aa0] [warning] No such vendor: LulzBot
[2021-11-04 18:26:51.831228] [0x00001aa0] [warning] No such vendor: PrusaResearc
h
[2021-11-04 18:26:51.831228] [0x00001aa0] [warning] No such vendor: TriLAB
[2021-11-04 18:26:51.831228] [0x00001aa0] [warning] No such vendor: Vertex
[2021-11-04 18:26:51.831228] [0x00001aa0] [warning] No such vendor: Voron
[2021-11-04 18:27:13.094865] [0x00001458] [info]    Trying to spawn a new slicer
 "E:\SuperSlicer\superslicer-gcodeviewer.exe"
Unhandled unknown exception; terminating the application.

supermerill, 2021-11-04:
zip is readable and correct for me, please retry.
And it's needed, as if you run something else, the log is useless.

knickohr, 2021-11-04:
[2021-11-04 20:19:43.513035] [0x00000684] [info]    Syncing configuration cache
[2021-11-04 20:19:43.513035] [0x00000684] [warning] No such vendor: Anycubic
[2021-11-04 20:19:43.528635] [0x00000684] [warning] No such vendor: Artillery
[2021-11-04 20:19:43.528635] [0x00000684] [warning] No such vendor: BIBO
[2021-11-04 20:19:43.528635] [0x00000684] [warning] No such vendor: Creality
[2021-11-04 20:19:43.544235] [0x00000684] [warning] No such vendor: FLSun
[2021-11-04 20:19:43.544235] [0x00000684] [warning] No such vendor: INAT
[2021-11-04 20:19:43.559835] [0x00000684] [warning] No such vendor: LulzBot
[2021-11-04 20:19:43.559835] [0x00000684] [warning] No such vendor: MakerGear
[2021-11-04 20:19:43.559835] [0x00000684] [warning] No such vendor: PrusaResearc
h
[2021-11-04 20:19:43.575435] [0x00000684] [warning] No such vendor: TriLAB
[2021-11-04 20:19:43.575435] [0x00000684] [warning] No such vendor: Vertex
[2021-11-04 20:19:43.591035] [0x00000684] [warning] No such vendor: Voron
[2021-11-04 20:20:04.107072] [0x00000dfc] [info]    Trying to spawn a new slicer
 "E:\SuperSlicer\superslicer-gcodeviewer.exe"
Unhandled unknown exception; terminating the application.
[2021-11-04 20:20:14.211090] [0x00000dfc] [debug]   boost::filesystem::permision
s before copy error message (this could be irrelevant message based on file syst
em): Das System kann die angegebene Datei nicht finden
[2021-11-04 20:20:14.242290] [0x00000dfc] [debug]   message handler shutdown().
[2021-11-04 20:20:14.257890] [0x00000dfc] [debug]   boost::filesystem::permision
s before copy error message (this could be irrelevant message based on file syst
em): Das System kann die angegebene Datei nicht finden

E:\SuperSlicer>

supermerill, 2021-11-05:
I'm sorry, can you test agin with https://github.com/supermerill/SuperSlicer/suites/4260394162/artifacts/110696472 ?

knickohr, 2021-11-05:
This debug version has some missing libraries, for example ucrtbased .dll and msvcp140d.dll, won't start.

supermerill, 2021-11-05:
weird, here the same zip with these dll from my system32.
https://mega.nz/file/N55liIhY#fXN9NwCbk33Myw2pFzABKYsXAy5nf1ABIDVGenpM--U

knickohr, 2021-11-07:
The latest debug version works without errors. Do you still need an console output ?

supermerill, 2021-11-07:
if it's fixed, it's all good.

closed 2021-11-19 by supermerill