supermerill/SuperSlicer #1780: Make center of rotation selectable

ristein, 2021-11-03:
**Is your feature request related to a problem? Please describe.**
When looking at some detail I want to rotate the view around this spot to get a better view of the 3d-geometry of lines. But when moving around one needs to be disciplined to have the center of rotation as needed. Otherwise, after zooming into that detail and rotating, the camera is somewhere completely else.

**Describe the solution you'd like**
By clicking with the middle mouse button somewhere on plater or some part, choose this point as center of rotation.

**Describe how it would work**
as above

**Describe alternatives you've considered**
none

supermerill, 2021-11-03:
It's a bit complicated, as you have to map a I/J into a X/Y/Z, so you need a plane where you stop your ray.
Maybe opening a dialog, where your can enter the new X,Y and Z coordinates where you want the new centre of view, opened via a keyboard shortcut?

ristein, 2021-11-03:
Hm, I hoped this was not that complicated since it detects parts already on mouseover and hoped clicking them could just deliver the coordinates.
I don't think that dialog would be more convenient than moving the CoR to the desired position by tilting 90° and repositioning.

neophyl, 2021-11-03:
This has been requested over on the Prusa Slicer github too.  
https://github.com/prusa3d/PrusaSlicer/issues/6805

ristein, 2021-11-04:
that Z command mentioned there (prusa3d#6805) only works in the 3D view tab, would be already a help if this was enabled in the other two tabs as well.
But when selecting an object in the list there is no indication which one it is on the gcode preview or sliced view, so still no good solution

supermerill, 2021-11-04:
> But when selecting an object in the list there is no indication which one it is on the gcode preview or sliced view, so still no good solution

you can do a quick ctrl+1 -> crtl+3 to select the one you want.