supermerill/SuperSlicer #1781: Crash or error on infill generation

TheBoojah, 2021-11-03:
I get an instant crash or some error message that varies, just after perimeter calculation is done and infill begins.

No problem on 2.3.57.3

[Error on infill.zip](https://github.com/supermerill/SuperSlicer/files/7467839/Error.on.infill.zip)

Version:   2.3.57.4
Build:     SuperSlicer_2.3.57.4_2021-11-02

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,272MB

supermerill, 2021-11-03:
Magnificent
[image]

neophyl, 2021-11-03:
Wow that looks like its exploding.

closed 2021-11-03 by TheBoojah