supermerill/SuperSlicer #1782: how to undo a merge

ristein, 2021-11-03:
**Describe the bug**
after merging parts I can't find a way to undo this

**To Reproduce**
Steps to reproduce the behavior:
1. Merge some parts
2. Try to unmerge

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
Some way to unmerge if one part should get 

**Screenshots**
Split to parts is something else I guess
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.57.3

supermerill, 2021-11-03:
only way: crtl+Z

supermerill, 2021-11-03:
for the red/green part, if you clic on the name (red part), you select the object. It's useful to rotate, delete them, etc...

ristein, 2021-11-03:
oh, I did not get that left right thing. Does not seem intuitive to me. Getting that popup also with a right mouse click on red would I find better. And a default column width like this:
[image]

supermerill, 2021-11-03:
> Getting that popup also with a right mouse click on red

Oh, the right clic? yes, why not.