supermerill/SuperSlicer #1784: Infill generated for areas that are not filled

srberard, 2021-11-03:
**Describe the bug**
SuperSlicer is generating infill incorrectly for a model where areas are solid where they should not be solid.  The very same STL sliced in Cura with similar settings did not exhibit this problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the [Parametric Hole Grommet & Snap-in Flush Cover](https://www.thingiverse.com/thing:1030514/) from Thingiverse 
2. Slice the "Parametric_Panel_Grommet_Flush_Outside.stl" file using SuperSlicer.  In my case I used the default printer settings for the VORON v2 with a 0.4mm nozzle.  I was able to reproduce this with the built in "0.2mm 0.4nozzle V2" print settings, "Basic PLA" filament settings, and 13% infill (though I was able to reproduce this with other settings as well).  
3. Examine the Gcode preview.  (the printed part is filled like the preview shows)

**>> Project File <<**
[ProjectFile.zip](https://github.com/supermerill/SuperSlicer/files/7469458/ProjectFile.zip)
Note:  the STL files are not mine, see the Thingiverse link for the full part files. 

**Expected behavior**
In this case, the center of the grommet should be left blank.   

**Screenshots**
Model in 3D View: 
[image]

The generated gcode is producing this:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 2.3.56

supermerill, 2021-11-03:
99% change that is a stl problem.

neophyl, 2021-11-03:
yeah fixing the file allows it to slice properly.  Yet another scad generated model that has errors.

Bet you will be glad when the PS2.4 merge is done as they have addressed this by changing the auto fix functionality to deal with the most common scad model errors.  Should have a lot less of these github issues posted after that :)

srberard, 2021-11-03:
I can confirm that doing a "Repair STL" in SuperSlicer fixed the issue.  I suspected it was an STL problem but as neither SuperSlicer or Cura flagged any issues so I just dismissed it.  My bad.

Closing issue.

closed 2021-11-03 by supermerill