supermerill/SuperSlicer #1785: Add other options to the quick settings section

Goodfeat, 2021-11-04:
**Is your feature request related to a problem? Please describe.**
In the slicer, you can have a very small list of quick settings, it would be nice if this list could be greatly expanded

**Describe the solution you'd like**
Is it possible to add additional parameters to quick settings?

**Describe how it would work**
Just click on the add a setting button, select the one you want from the list, and the setting appears in the quick settings menu.
[image]

**Describe alternatives you've considered**
an alternative addition can be implemented in the settings menu, for example, click on the icon next to the desired setting, and this setting will appear in the quick settings menu
[image]

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2021-11-04:
It's very hardcoded, and the brim and support one are special, as it's not a real setting, but a placeholder that will then modify the setting.
These kind of "easy mode setting" are interesting, even more so for a "very easy mode". Maybe I should make more of them. But maybe not before a easy-expert tag rework.

If I modify the code to let the user modify this section, it will be via a ui file like for the rest of the setting ui.

Goodfeat, 2021-11-04:
Yes, I understand you very well. I noticed that this section does not change in any way from version to version of the slicer, although there are other settings that I would like to change immediately.

  I will be grateful if you have time and you can finally modify the code to expand this section to add other settings.

supermerill, 2022-07-12:
will update the readme in resources/ui_layout.
For now, check my awnser in 2897

closed 2022-07-14 by supermerill