supermerill/SuperSlicer #1786: Allow any setting on a printer to be linked to the filament used

stewcop23, 2021-11-04:
I have recently gotten started with printing TPU, and am having some stringing problems, which is to be expected. I exchange the filament regularly on my printer, and would not like to have to change the "avoid crossing perimeters" setting every time I change filament.

I would like the ability to right-click (or another shortcut) a setting (in the print or printer menus) and link it to a specific filament. that way, I would have to hunt through menus to find the settings that I want to turn on and off for different filaments

I hope I've made this idea clear enough, if not I would be happy to try and describe it further!

supermerill, 2021-11-04:
isn't it what the "dependencies" tab is for?

stewcop23, 2021-11-04:
I haven't fully explored the dependencies tab (as this is my first time hearing about it, also how do you use it?) so I can't be fully sure but, to my knowledge, this tab doesn't actually change the parameter, it just stops the profile from being used if it doesn't contain it. if I'm wrong then this suggestion can be removed, but the dependencies interface is still terrible and could use some improvements.

neophyl, 2021-11-04:
Sounds like they want every setting from print and printer profiles available as an over ride on the filaments profile. I’d imagine that would take some major refactorin.

supermerill, 2021-11-04:
mmm that seems a bit difficult to implement.
first, settings that are global to the plate can't be overridden, as it create all sort of issues with multiple filaments.
Other than that, it should be possible, but the framework isn't here.