supermerill/SuperSlicer #1788: Errant mouse

bigalgeorge, 2021-11-04:
Open up SS (single instantiation)

About 50% time on my Ubuntu/Budgie if I left-click and move mouse the platen does not rotate (desired method)

When this situation, I found if I go to Config/Preferences, and dont do anything, exit, then the mouse will behave as expected (rotational)

This can happen at any time and without loading a Project.
**>> Project File <<**

If required, I can make a short video when problem happens again
[image]
?

supermerill, 2021-11-04:
plater rotation should happen even if you press the left button over anything else, as the plater should take the focus every time the mouse hovers it. (and I found that a bit irritating when I try to modify something on the right panel)

Maybe on ubuntu it's different, will test.

bigalgeorge, 2021-11-04:
When I got time will do a short vid.
I think its likely uncommon, and possibly is not worth considering unless others are reporting similar issues.

closed 2021-11-05 by bigalgeorge