supermerill/SuperSlicer #1791: Support In Mid Air/Thru Model on Difficult Geometry

kpdcompany, 2021-11-04:
**Describe the bug**
I have repaired and repaired and repaired, tried multiple facet settings from course to fine, but this curve/arch on these parts is throwing the interface support and Support Enforcers for a loop.  If for some reason I am an idiot and this is something else, please advise.  When doing "Support first "X" Layers, anything 6 Layers or less, works without this issue, when increasing or using a support slab, it gets that random little extrusion out in the middle of nowhere.  This is something I noticed on the last release too, so I am pretty sure it is a Bug as S3D prints and supports perfectly fine.  I look forward to tracking this down and joining the community.  Thanks Guys in advance! 

**To Reproduce**
Steps to reproduce the behavior:
1. Load My Project
2. Click on 'Slice'
3. Look at ends of part to see random air support.
4. You can turn off a slab, or First Layer support and it goes away, but so does my support where I need it, so this is currently not even printable with SuperSlicer.

**>> Project File <<**
Attached

**Expected behavior**
I would like to get this issue fixed and test possible fixes ASAP, these are production parts and I am stuck using S3D right now.  Any help would be awesome.

**Screenshots**
Attached

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 10.0.19042 Build 19042

[KPD Support Issue.zip](https://github.com/supermerill/SuperSlicer/files/7475222/KPD.Support.Issue.zip)

5axes, 2021-11-04:
Don't use the sawtooth Pattern.  
[image]

I even don't know what kind of benefit we can have with this type of pattern ????

[image]

Same type of remark concerning the black bed for the Voron machine ? not easy to analyse the result would be better to keep a ligth design ?

[image]

kpdcompany, 2021-11-04:
Thats weird, I dont even use the voron backgound/bed, see my screenshots.

As far as Sawtooth, this was the only way I was getting good bridge surfaces.  This is a GF Nylon part so bridging that arch is pretty important.

But THANK YOU, this narrowed it down, it is only WITH Sawtooth Interface.  Thank you!!! 

On that note, we should be able to get it to work right?  But this is less of a pressing issue now.

kpdcompany, 2021-11-04:
any ideas on this?  Keep getting support where I dont want or nee it, auto is off, using enforcers and I even painted it blocked, and still does it
[image]
.

5axes, 2021-11-04:
Well in this case I could suggest to rotate your model of 90° and reset to zero the support  material angle as a WORKAROUND TEMPORARY solution (as I suppose you don't want also to change the support direction) .
[image]

supermerill, 2021-11-04:
>  we should be able to get it to work right? 

It sounds like a bug, I'll check

kpdcompany, 2021-11-04:
> > we should be able to get it to work right?
> 
> It sounds like a bug, I'll check

Thank you, while you are at it, the Interface Layers also shoot out some even with regular settings, if you look at the latest pic, you can see on the tight and left interface layers jutting into mid air which could cause a print failure, shouldn't the Interface layers be bound by the support boundary?
[image]

kpdcompany, 2021-11-04:
> Well in this case I could suggest to rotate your model of 90° and reset to zero the support material angle as a WORKAROUND TEMPORARY solution (as I suppose you don't want also to change the support direction) . [image]

Thank you, this DID work, even for Sawtooth, so must be a bug.  Have you ever used Sawtooth?  It works really nice for stuff like this with really shallow angles.

kpdcompany, 2021-11-04:
keep finding bugs.  This support seems to be having issues.  When going "from filament" to "by plane" seems to be leaving either too much gap or not enough, respectively.  There is only a setting of a .15mm gap with .1mm layers.  From filament leaves it looking like the below where the outside single perimeter has support, but the rest has at least 2-3 layers of a gap, so support falls.  If using form plane the support melts into the bridge by over half the bridge. 
[image]
[image]
[image]

supermerill, 2021-11-04:
that's because your overhangs have no bridge flow. You have to activate it.

kpdcompany, 2021-11-04:
I have it set at 75%

supermerill, 2021-11-04:
weird, they are flat in the picture.

kpdcompany, 2021-11-04:
exactly my point, haha, I just closed out and went back in, and now it shows correctly.  Its just a factor of WHEN it starts, not flow % correct?

supermerill, 2021-11-04:
For the support, you set teh offset between the interface & the bridges. This offset can be computed from the bottom of the extrusion (from filament) or from the bottom of the layer (from layer). If the overhangs don't have the bridge flow, the computed height is wrong.

kpdcompany, 2021-11-05:
Thanks!  is there a way to fill the thin gap between the bridge perimeters and the bridge middle fill?  I appreciate your help!
[image]

kpdcompany, 2021-11-05:
so the above gap is DEFINETLY a bug, when I turned on overlap perimeters, it closed that gap, I uncheck it, then opens the gap.  I went back to check it, and then the gap did NOT close?  Somehow the perimeter overlap is not updating easily when checked/unchecked.  It is unpredictable

supermerill, 2021-11-05:
> there a way to fill the thin gap between the

There's not enough empty space to fill this.

You're not the first to ask for that, there are already multiple issues on that.

kpdcompany, 2021-11-05:
> > there a way to fill the thin gap between the
> 
> There's not enough empty space to add another line. You can play with bridge flow or bridge overlap, but that may create the same thing in another place. You're not the first to ask for that, there are already multiple issues on that.

Ok good to hear, an option for "Space between bridges and Overhangs" would be amazing, it would fix the support issues I am having.  Can't get the sweet spot without some way to control the width of the bridge separate from overlap and Flow, or maybe a "Bridge Gap Fill" Option that uses the bridge flow settings and dials back for the gap just like a thin wall.  Once I get it to the right flow and lap to make a tight bridge, its too over extruded and loops between supports.  We consistently run support models if you want someone to test that runs a ton of random model shapes, we can help out.

Thanks!

closed 2021-11-19 by supermerill