supermerill/SuperSlicer #1792: Color change added between layer 1 and 2

CDFI-3d, 2021-11-04:
**Describe the bug**
When i slice my model superslicer add a Color change command at the final of layer 1.

I have 1 extruder 1 material 

**To Reproduce**
Steps to reproduce the behavior:
1. Slice a model
2. edit the Gcode 
3. look the M600 at the final of layer 1
4. See error

**>> Project File <<**

[Problem.zip](https://github.com/supermerill/SuperSlicer/files/7476447/Problem.zip)

**Expected behavior**
i hope dont need this M600 at the layer change

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.57.3

supermerill, 2021-11-04:
I need your project file: `File -> save as` zip it and drop it here.

CDFI-3d, 2021-11-04:
[Problem.zip](https://github.com/supermerill/SuperSlicer/files/7476538/Problem.zip)

CDFI-3d, 2021-11-04:
sorry i forgot the project i do it

Pavulon87, 2021-11-04:
[image]

CDFI-3d, 2021-11-04:
ohhh i'm a fucking newb ..... thx !!!!! dont se it ....

supermerill, 2021-11-04:
in the vertical scrollbar, go to layer 2 and remove the colorchange you added by mistake by clicking on the (X)

closed 2021-11-04 by supermerill