supermerill/SuperSlicer #1793: 'Avoid crossing perimeters' doesn't work with 'Minimum travel after retraction'

d-schmidt, 2021-11-04:
**Describe the bug**
It seems it only works for retracted travel moves. I hoped it would simulate Curas "Combing"

**To Reproduce**
Steps to reproduce the behavior:
1. Have a small part. 10 mm 'Minimum travel after retraction' and 'Avoid crossing perimeters'

**Screenshots**
Minimum travel after retraction: 3 mm
[image]
Minimum travel after retraction: 10 mm  
oozes along the pink arrow while moving directly from red to red:
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version c50452a

supermerill, 2021-11-04:
Yes, currently it's only doing it if it's long enough to warrant it. When the travel is very short, like between two adjacent perimeters, there isn't much need to spend CPU time to see if it needs to do something smart.

I can add a hardcoded threshold...

d-schmidt, 2021-11-04:
The first screenshot shows it is long enough to warrant it. The only difference between screenshot one and two is the 'Minimum travel after retraction' distance.

d-schmidt, 2021-11-14:
can corfirm both options do now work together in fixed in 2.3.57.5

closed 2021-11-14 by d-schmidt