supermerill/SuperSlicer #1794: Avoid crossing perimeters - avoid moving along outer perimeter

jwalker55, 2021-11-04:
**Is your feature request related to a problem? Please describe.**
When using "Avoid crossing perimeters", it tends to want to follow on top of the outer perimeter. With bowden tools (and some more ooze-prone filaments), this just causes the oozing to deposit directly on the outer perimeter, which leaves an ugly surface finish. 

**Describe the solution you'd like**
It would be nice if the travel move could happen along one of the inner perimeters and not the outer.

**Describe how it would work**
As shown below, the red circle shows where the original travel move is placed using the "Avoid crossing perimeters" features, and my horribly drawn bright blue line represents where I'd like the travel move to be.
[image]

**Describe alternatives you've considered**
It seems a lot of other slicers do this same thing, which is kind of frustrating. Sometimes, I just disable "Avoid crossing perimeters" and pick the zits off the part where it crosses. Or, if the part allows, I adjust the settings so there are shorter/less travels (z seam, etc).

supermerill, 2021-11-04:
It would be nice. I tried to do that but it's a kind of big task to do that reliably.
I have a prototype that kinda work, but it very suboptimal. I'll have to scrap it to retry with another algo to do it prroperly.

jwalker55, 2021-11-04:
Understood. Thanks for the insight!

CCS86, 2021-11-05:
Yup, I asked for this a while back: https://github.com/supermerill/SuperSlicer/issues/1349

The current method really destroys the wall surface.

supermerill, 2021-11-05:
You may have an updated one for this we, if I can understand why some are still not taken into account.
[image]

CCS86, 2021-11-07:
New release is big!

jwalker55, 2021-11-08:
Awesome! Below is a real-world example of the improvement
[image]

closed 2021-11-19 by supermerill