supermerill/SuperSlicer #1801: Config Wizard - Custom Printer

bigalgeorge, 2021-11-05:
My 2nd printer is Voxelab Aquila X2 of which there is no such in printers.
Using the Config Wizard has some non-expected logic:

Config Wizard/Custom/Bowden .4/Next...............screen full of Creality printers
Expected input details for custom printer.
If [Next] is pressed till Custom Printer then its correct; it should automatically skip to this item.

supermerill, 2021-11-05:
I don't understand exactly.
If I want to create a custom printer from the wizzard, I start here:
[image]

then, clicking next: I go to firmware, bed, temp
and then it goes back to the standards tab: [filaments if any vendor selected], updates, reload from disk, files associations, view mode.

i didn't see any creality printer.

bigalgeorge, 2021-11-05:
It was my confusion, but could effect others I thought selecting 'Custom' would define the feed for the new printer.
Not so, its a separate setting, as Feeder. Perhaps changing the name from 'Custom' to 'Custom Feed' would help other Newbies?

closed 2021-11-05 by bigalgeorge

supermerill, 2021-11-06:
What's a 'Feed' ?

bigalgeorge, 2021-11-06:
EG Bowden tube