supermerill/SuperSlicer #1802: retracts do not work in the modifier

Goodfeat, 2021-11-06:
**Describe the bug**
Hello, I tried to adjust the retracts using the layer modifier, but the retracts do not work at all

**To Reproduce**
Steps to reproduce the behavior:
1 create a range of layers
2 add a retract modifier to each range
3 send to print to calibrate retracts
4 the model is printed, but the retracts do not work

**>> Project File <<**
[retract_calibration_towers_6112021_1853_SSli3r2.3.57.zip](https://github.com/supermerill/SuperSlicer/files/7490149/retract_calibration_towers_6112021_1853_SSli3r2.3.57.zip)

**Expected behavior**
I want to get a different retract at each interval of the part

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [e.g. 2.3.57.3], Use `Help->About`_

**Additional context**
Add any other context about the problem here.

neophyl, 2021-11-06:
Your zip just contains gcode.  Merill will need your saved **Project**.  File>Save project as.  This will save a 3mf with all your settings, the modifiers and model etc.

Goodfeat, 2021-11-06:
[retract_calibration_towers3fm.zip](https://github.com/supermerill/SuperSlicer/files/7490272/retract_calibration_towers3fm.zip)

neophyl, 2021-11-06:
The values you have set in your modifiers should not be negative values.  If you want a retaction of 5mm then you need to set 5, not -5.  Remove the negatives and it works.

Goodfeat, 2021-11-06:
Sorry, I was sure that it was necessary to put a negative value, since the value was specified in the example -1

neophyl, 2021-11-06:
That's not an example.  -1 is the default value.  If you hover over until you get the tooltip then it actually says set negative to disable.  So the default is disabled.

Goodfeat, 2021-11-06:
Thank you very much, I did not study the modifier well, I will try to be more attentive

closed 2021-11-06 by Goodfeat