supermerill/SuperSlicer #1804: "Only one perimeter" setting causes overlapping infill w/ perimeters when there is no room for top infill

scottmudge, 2021-11-06:
**Describe the bug**
On smaller objects, the "Only one perimeter" setting causes overlapping infill with perimeters.

[image]

**To Reproduce**
Use default settings (0.20 Fast), slice attached model.

**>> Project File <<**

[example_issue.zip](https://github.com/supermerill/SuperSlicer/files/7490422/example_issue.zip)

**Expected behavior**
I would expect that in cases where there is no room for top-layer infill (completely filled by perimeters), the infill would be disabled. 

**Screenshots**
See above.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: Latest nightly (2.3.57.4)

**Additional context**
Disabling "Only one perimeter" corrects the issue.

supermerill, 2021-11-06:
haha, yes. I also found it and fix it an hour after the nightly

neophyl, 2021-11-06:
Looks ok in the previous one 2.3.57.3 so must be a change in the nightly.

scottmudge, 2021-11-06:
Great, I'll close this issue. If I see it again I'll make a note here. Thanks!

closed 2021-11-06 by scottmudge