supermerill/SuperSlicer #1811: Building SuperSlicer on RPi4 Buster 32Bit.

MrRKernelPanic, 2021-11-06:
### Operating system type + version
Rasberry Pi OS (Buster) 32-Bit - 8GB Pi4 @2.3GHz from SSD.

### Behavior
* Problems when building, Error 2 and missing wxWidgets.
* Try running guided instructions:
* Installed Dependencies.
sudo apt-get install -y git cmake libboost-dev libboost-regex-dev libboost-filesystem-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libcurl4-openssl-dev libwxgtk3.0-dev build-essential pkg-config libtbb-dev zlib1g-dev libcereal-dev libeigen3-dev libnlopt-cxx-dev libudev-dev libopenvdb-dev libboost-iostreams-dev libnlopt-dev libdbus-1-dev

Cloned the GIT.
make a build directory and navigated into it.
cmake .. -DCMAKE_BUILD_TYPE=Release -DSLIC3R_BUILD_TESTS=OFF -DSLIC3R_GTK=3

First error I came across was regarding wxWidgets (the Pi seems to get confused between GTK2 and 3, I tried adding this switch which seemed to get me a step further.)

-DwxWidgets_CONFIG_EXECUTABLE=/usr/lib/arm-linux-gnueabihf/wx/config/gtk3-unicode-3.0

But I seemed to get this error no matter what I tried.
make[1]: *** [CMakeFiles/Makefile2:1280: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

I went back and made sure that all the dependencies were build, there were no errors.  But it would just not build.

Eventually out of frustration I tried using Ninja.
- checked the dependencies 
- (navigated the deps folder, 
mkdir build && cd build
cmake -G Ninja ..
Ninja
(This again build with no errors)

-Navigated to the root of SuperSlicer, then repeated, 
mkdir build && cd build
cmake .. -G Ninja -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"
Ninja
Ninja install

Built first time, no errors and works a treat.  I'm running this on Pi4 8GB Overclocked to 2.3GHz (Argon Case) running from SSD using /usr/local/bin/superslicer

As a long time user of PrusaSlicer on the Pi4 this come with lots of added benefit and support Kipper which will be my next upgrade.

supermerill, 2021-11-07:
Nice.

Is there a problem? I didn't see any question.
You didn't try the BuildLinux script? I'm wondering if it could have works. But whatever, if ninja is simpler and more effective.

for reference: #486

MrRKernelPanic, 2021-11-07:
Thanks for the reply.  I did try all those steps, but couldn't seem to get
the right combination, the final Make/CMake command always failed.  I spent
a long time going through both posts on this Github and also on the
PrusaSlicer for the Pi.

Just for your information.

Keep up the excellent work, my 8 year old daughter is also amused as
SuperSlice has support for 'Mickey Mouse Ears', which we have needed in the
past for curling corners and for prints without a heated bed.

Kind regards

Mark.
_____________________________________________________________

"No trees were destroyed in the sending of this message. However,
a large number of electrons were terribly inconvenienced."