supermerill/SuperSlicer #1813: One part affecting top layer another

Pavulon87, 2021-11-06:
**Describe the bug**
Why is one part changing top layer of another when it shouldn't? Interface shells turned on. Prusa slicer leaves top layer clean. Susi leaves a trace of the top object. Is there any option to avoid this?

**To Reproduce**
Steps to reproduce the behavior:
1. Import/create multipart object
2. Place one part just above the other
3. Turn interface shells on
4. Slice
5. See error

**>> Project File <<**
[toplayer.zip](https://github.com/supermerill/SuperSlicer/files/7491143/toplayer.zip)

**Expected behavior**
Leave the top layer unchanged

**Screenshots**
Susi top:
[image]

Prusa top:
[image]

Susi top-1:
[image]

Susi's weird artifacts:
[image]

Prusa:
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.57.3

supermerill, 2021-11-07:
For the first object, it's because you ask for 3 periemters and only 1 visible on top surfaces. That's what it does.
I don't see any problem here.
[image]

For the second object, i can't test. Maybe the same thing.

Pavulon87, 2021-11-07:
So what is the purpose of this? Why can't the top layer of square remain unchanged? Better adhesion?
I can change this by inserting the top object as a separate body. But there is a problem to put the second object at the correct height, because the z position is automatically reset if there is only one part.

supermerill, 2021-11-07:
you asked for 3 perimeters, why remove all of these because some of the layers has a top surface?
you can use a range modifier to modify the perimeter count to 1 if this is what you want.

example where it's (more?) needed
[image]

Pavulon87, 2021-11-07:
Maybe I didn't make myself clear. I asked why the top of the square had a circle on it, while the Prusa slicer leaves a smooth surface. It's not about the number of perimeters on the top layer. I don't see why it should work that way because it causes imperfections. 
But never mind, I found a workaround.
[image]

[toplayer.zip](https://github.com/supermerill/SuperSlicer/files/7493047/toplayer.zip)
I think the issue can be closed.

supermerill, 2021-11-07:
> Maybe I didn't make myself clear. I asked why the top of the square had a circle on it, while the Prusa slicer leaves a smooth surface. It's not about the number of perimeters on the top layer. I don't see why it should work that way because it causes imperfections.

oh, it's because on prusaslcier, there is a hardcoded 3mm "anchor". In susi, it's available in "external_infill_margin" and default to 150% (so ~0.6mm).

closed 2021-11-08 by Pavulon87