supermerill/SuperSlicer #1814: FR: Retract Continue in Travel

CCS86, 2021-11-07:
The idea here is to optimize the retraction length, based on how long of a travel move is next. 

You would have have an initial retract move performed at the end of the print segment, then a simultaneous retract-travel move, possibly followed by a pure travel move.

The premise is that a longer travel move has a higher likelihood of oozing, and therefore could use a longer retract move to prevent the ooze. A potential secondary benefit is that some of the retraction would happen during the travel move, preventing more ooze by occurring over a longer duration. By reducing retraction distance on shorter moves, the print is more efficient.

You could define it in 3 parameters:

*Initial retract distance (mm)
*Continuation factor (mm/mm) (a decimal multiplied against the length of the travel move in mm)
*Maximum retraction distance (mm)

There is a Cura plugin for this function that works quite well.

supermerill, 2021-11-07:
like "Retract amount before wipe", but "Retract amount before travel"?

CCS86, 2021-11-07:
Yeah pretty much. But, with the added control of the retraction velocity within the travel move. This also ends up effectively setting the travel distance to reach full retraction. I suppose the existing retract distance could act as the maximum retraction distance.

supermerill, 2021-11-07:
so more like a "continuous retraction on travel at speed:" XX "mm/s"
and all that will be unretracted on arrival?

CCS86, 2021-11-08:
Maybe this chart will explain more. Two different hypothetical settings.

Yes, everything gets un-retracted at the end of travel.

[image]

mrmccain9, 2021-11-12:
“Retract Continue” is honestly the biggest thing I miss from Cura. It’s surprising that it wasn’t more widely known.

It was great because I had it set to a ratio of 0.03, so for example a travel move of 20mm would retract 0.6mm, relatively quickly, but then a 200mm travel (torture test) would retract 6mm with no oozing (PETG). It really took my print quality to the next level in terms of not leaving zits and holes after travel moves. It was great for speed too, since small travels, like 5mm between infill areas for example, would only retract 0.15mm. Worked well for single parts or for a bed full of parts.

That said, I’m loving superslicer and don’t see myself going back to cura. Keep up the great work!

rtm7777, 2023-04-28:
I'd love to see this implemented, right now I forced to change retration length for small parts to get some useful result
But when I set high number of instances seams looks perfect because of long travel moves. But for single part they look ugly

djessup, 2023-05-12:
I've often wondered about something quite similar to this, with a small but significant distinction. 

The idea I'm alluding to is **combining** a portion of a retraction with the following travel move - for the purpose of discussion, imagine you have a default retraction of 2mm configured and the toolhead is about to make a 100mm travel move.

The slicer could make an initial retraction move - say 1mm (50% of 2mm) - in the same stationary manner it currently does, then perform the G0/G1 travel move of 100mm, while _also_ retracting the remaining 1mm, doing it over the course of the 100mm travel move. 

The hypothesis being tested here is that an initial fast retraction creates low pressure spike in the nozzle, then maintains relative low pressure with a longer, slower retraction during travel. 

The high viscosity would act as a buffer of sorts (in my head it's a bit like an electric capacitor, charged by the retraction and discharged in the form of ooze.. I can torture this analogy more but hopefully you get the idea 😉) and with a little tuning I hope this could both improve print quality/accuracy, and possibly be slightly faster, since less time is spent with the toolhead stationary.

Some ideas as to what inputs this feature might take -
* Target retraction length (the total amount of retraction to make, both while stationary and travelling - could reuse existing retraction length setting)
* Stationary retraction amount (% of total retraction, or absolute retraction in mm, that should be done while stationary. Assume remainder is done during travel)
* Stationary retraction speed (mm/s speed of retraction while stationary, could reuse existing retraction speed setting)
(Travel retraction speed can be calculated based on the amount of retraction and length/speed of the travel move.)

There are other aspects that could be configurable, like ending the travel retraction slightly before the end of the travel move (essentially splitting it into two portions, travel+retract and just travel) so the pressure can start normalising slightly before the next extrude move. 

I also like the idea of variable retraction length based on the length of the travel move, and I think these two concepts could marry quite nicely together, so rather than a "target retraction length" you might have "initial retraction length" and "travel retraction ratio" that sets the magnitude of retraction per mm or duration of travel.

If it's feasible I might try playing with a post-processing script to experiment with this concept, and I'm keen to hear from anyone else who has had similar ideas, or better yet has any real-world experience/data how well such a feature works in practice.

CCS86, 2023-05-12:
> I've often wondered about something quite similar to this, with a small but significant distinction.
>
> The idea I'm alluding to is **combining** a portion of a retraction with the following travel [...]

I don't see how that is different than my original request.

djessup, 2023-05-14:
> > <mi-derpa>
> I don't see how that is different than my original request.

My apologies, when I first read over your post I thought you were asking for retraction length based on the length of the following travel move, but essentially with the retraction sequence the same as it currently is - retract, _then_ move.

Looking again I see that you indeed mention combining some of the retraction with the travel 🫠 - so clearly that's my error, I'm sorry for that!