supermerill/SuperSlicer #1815: Error when running calibration tests on NAS storage

Goodfeat, 2021-11-07:
**Describe the bug**
Hello,
Found a bug when using the program via NAS storage

**To Reproduce**
Steps to reproduce the behavior:
1 The program to the NAS storage
2 Run the slicer program
3 Go to the calibration section
4 Trying to call any type of calibration (for example a calibration cube)
5 I get the error
 
```
  SuperSlicer Error
    Unable to open requested HTML document: \\Nas\Р°СЂСРёРІ_РїСЂРѕРіСЂР°РјРј\SuperSlicer_2.3.57.4_S\resources/calibration/cube/cube.html
```

**Expected behavior**
Waiting for the calibration settings page to launch

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [e.g. 2.3.57.4], Use `Help->About`_

**Additional context**
the slicer is probably not designed for work via NAS storage, but all the main functions of the program work well.

Most likely, the program cannot find the path correctly, although everything works without problems on a computer and USB media

supermerill, 2021-11-07:
Can you test this version, (it contains some changes)?
https://mega.nz/file/hoJAQBhS#JiPJNAOQ4j-TKUIkZ85j6mIrX5tbxoXCbRYGjHvUNCY

Goodfeat, 2021-11-07:
[image]
No, this also does not work unfortunately

supermerill, 2021-11-07:
mmm... maybe an Unicode issue.

Goodfeat, 2021-11-07:
it seems the program does not work with Cyrillic

[image]

supermerill, 2021-11-07:
I tried to reproduce but I can't. maybe because I set my windows to work with uft8...

Goodfeat, 2021-11-07:
It seems I have windows encoding in cp-1251

[image]

Goodfeat, 2021-11-07:
I tried to install in windows using unicode.  On windows, this mode is in beta status.  Most likely bugs will appear in other programs

[image]

Goodfeat, 2021-11-07:
[image]

supermerill, 2021-11-07:
Can you test this one? only the calibration cube.
if the error message is different, show me the new one.

Goodfeat, 2021-11-07:
i just checked out a new build
now everything works correctly,

bug has been fixed

[image]

closed 2021-11-19 by supermerill