supermerill/SuperSlicer #1826: Slicing that does not progress beyond "Fill Generation"

macchinette, 2021-11-08:
**Describe the bug**
When slicing the stl the slicer stops on "Fill generation".

**To Reproduce**
Steps to reproduce the behavior:
1. Open attached 3mf
2. Slice file
3. GUI still works
4. Press cancel
5. program freeze

Check:
1. task kill slicer
2. open file again
3. change the value of "bridge_angle" other than "0"
4. slice successful

**>> Project File <<**

[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/7498868/3DBenchy.zip)

**Expected behavior**
Slice successful any value.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version SuperSlicer_2.3.57.5_2021-11-07

**Additional context**
I am not a great expert, and I was wondering how the automatic calculation of the angle of the bridge works, I thought that the best direction should be in the case of 4 walls so a rectangular area (such as the top of the boat cabin) the shortest direction between the walls.
I have seen that often when it is automatically calculated the direction depends on the rotation of the piece and is not distributed in the shorter direction.

supermerill, 2021-11-08:
I think it's the bridge bug I found yesterday. Will do a release for it.

macchinette, 2021-11-08:
> Penso che sia il bug del ponte che ho introdotto e trovato ieri. Farà un rilascio per questo.

Hello Supermerill,
yes the problem was that and I also noticed that in my humble opinion the direction of the bridge is better calculated.
Thanks

closed 2021-11-19 by supermerill