supermerill/SuperSlicer #1829: Can't Slice any file, error message about "machine_max_acceleration_X"

orecus, 2021-11-08:
### Version
Version 2.3.57

### Operating system type + version
Windows 11, Latest Updates

### Behavior
Error message saying: 
`You can't define an option that need machine_max_acceleration_X without defining it!`

I'm using Klipper, and SuperSlicer is configured as using that, if I change to Marlin I still get the same problem though. Guessing it's something with my profiles that clashes with the latest version, they worked fine in the previous build though.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Can't export the file as it crashes due to the above error.

supermerill, 2021-11-08:
I updated some acceleration fields to be able to be a % machine_max_acceleration_X. But these interactions may creates some bugs when the value isn't here.
can you share your project file for helping me to fix the bug?
(file -> save as, zip, drop it here)

orecus, 2021-11-08:
@supermerill Can't save it, due to the above error, Superslicer forcefully quits after trying. There is a file generated, but I don't know what state it's in though. Hope it helps!

[xyzCalibration_cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7499532/xyzCalibration_cube.3mf.zip)

supermerill, 2021-11-08:
then, can you export the config? (file -> export -> export config) ?

orecus, 2021-11-08:
Sure, here it is.

[SuperSlicer - FLSUN SR.ini.zip](https://github.com/supermerill/SuperSlicer/files/7499568/SuperSlicer.-.FLSUN.SR.ini.zip)

supermerill, 2021-11-08:
can't open any of your zip files.

orecus, 2021-11-08:
@supermerill they are just renamed files, so remove .zip from the name and you should be able to use them.

supermerill, 2021-11-08:
oh, ok

eduncan911, 2021-11-14:
I can confirm that upgrading from 2.3.57.4 to 2.3.57.5 fixed this error, at least for me.

closed 2021-11-19 by supermerill

chrzr, 2021-11-29:
Hi,
I also just recently ran into this issue after migrating to a new Superslicer version.
The error is related to Acceleration Control options under advanced and it seems to be triggered, if the default value is expressed as 0%.

[image]

With this configuration I am always running into the same error when trying to slice any model.

I am currently using version 2.3.57.4