supermerill/SuperSlicer #1830: Calibration: Flow Error "some Objects are too close"

inspectionsbybob, 2021-11-08:
2.3.56

### Operating system type + version
Win10

### Behavior
Trying to Tube my Printer by going through the calibration routine. Got this on the Flow Calibration

JGMaker Artist-D, 310x310x350 bed, two extruders. heated bed, 

[image]

supermerill, 2021-11-08:
you should be able to space them by clicking on the "arrange" button.
but it should be already done.
can you save the project, when it refuses to slice, zip it and drop it here? I can't reproduce the issue with my settings.

inspectionsbybob, 2021-11-08:
Here you go

supermerill, 2021-11-08:
? 
There is nothing?
did you sent it by email instead of posting it on github?

inspectionsbybob, 2021-11-08:
Lets try again with GitHu
[Flow calibration.zip](https://github.com/supermerill/SuperSlicer/files/7500590/Flow.calibration.zip)
b

supermerill, 2021-11-08:
fixed in the latest beta release.

inspectionsbybob, 2021-11-08:
Ok, I got it to export to a different printer after moving the blocks around.  My Printer really didn't like the file.  It made a very ugly sound and shook like crazy as if it was stuck but wasn't.  I aborted the print on the third block after I took the video...   (The -20 looked the best of what was printed)

[IMG_7590.zip](https://github.com/supermerill/SuperSlicer/files/7500778/IMG_7590.zip)

closed 2021-11-19 by supermerill