supermerill/SuperSlicer #1831: Height range modifiers get overriden

nfons, 2021-11-08:
**Describe the bug**
previously

**To Reproduce**
Steps to reproduce the behavior:
Create 2 boxes, set one to z height 14 (z14), and another to 15 (z15).

modify the z height 15 to add height range modifier of top layers of 10.

Notice how when both the z14 and z15 cubes are made, they both begin to fill top layers at relatively same rate from 13-15 layer height.

now add new height range mod on z15 cube from 14.6 - 15.0 and add 0 top layers, 0 infil.

notice how the top layers (13-14.3) are no longer created.

Compared with prusaslicer 
[image]

where it works as intended

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

normal:

[image]

with top gone:

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 2.3.55.5], Use `Help->About`_

**Additional context**
Add any other context about the problem here.

nfons, 2021-11-08:
Note: this is NOT an issue on 2.3.56

supermerill, 2021-11-08:
[image]
[1831_whatis_the_pb.zip](https://github.com/supermerill/SuperSlicer/files/7499757/1831_whatis_the_pb.zip)

I fail to see or understand the problem.
Maybe if you send me your problematic project file?

nfons, 2021-11-08:
@supermerill it shouldnt look like that...it should look like this:

[image]

note the top surfaces EXCEPT for last 2 layers (which has only perims)

supermerill, 2021-11-08:
can you save your project file and post the zip here?

nfons, 2021-11-08:
[14-full.zip](https://github.com/supermerill/SuperSlicer/files/7499887/14-full.zip)

Start this up.  I have colored for better visual.

Notice top layers start at z12.2

Now add new height mod from 14.6-15, add 0 top 0 infil (essentially a hollow square for JUST the 2 top layers)

Notice now that top layer does not start from z12.2. thus, all the mods from 12.2-14.6 is ignored

previous version (2.3.56.6) does not have this issue

supermerill, 2021-11-08:
strange side effect of "no solid infill over X perimeters"

supermerill, 2021-11-08:
If you remove "ensure vertical shell thickness", all these solid layers are removed.

I think I need to study exactly what's doing what, with and without "ensure vertical shell thickness", before making more changes in this part of the slicing algo.

For the moment, I will say that it's the effect of "no solid infill over X perimeters" and there isn't an obvious "solution".

there is also the problem of the first layer modifier breaking bottom count... and things like that.

nfons, 2021-11-09:
@supermerill yea, setting the no-infil value to higher z than the model, "fixes" it