supermerill/SuperSlicer #1832: Latest pre-release does not start, stable does start

kursatu, 2021-11-08:
**Describe the bug**
I've upgraded to windows 11 recently. 
I';ve got this and it does not work. 
https://github.com/supermerill/SuperSlicer/releases/tag/2.3.57.4
This however works.
I've also had my build and I've recently pulled the sources and ibuilt..
It also stopped working.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 2.3.55.5], Use `Help->About`_

**Additional context**
Add any other context about the problem here.

supermerill, 2021-11-08:
Just launching the slicer with the exe?
if you launch it with an empty configuration directory (with --datadir setting or with an empty "configuration" directory next to the exe), does it launch or crash instantly?
if you launch the '_console' one from a command prompt, does it output an error message?

kursatu, 2021-11-08:
- Just launching the superslicer.exe.
- With -datadir to an empty directory next to the exe,  it works
- superslicer_console.exe starts and exits after splash screen and it prints the following
create_preset_tabs: 3
update_layout1: 3
I can attach my current datadir if it helps.

supermerill, 2021-11-08:
> I can attach my current datadir if it helps.

yes, please

supermerill, 2021-11-09:
also, if you can't upload your configuration folder, can you test the latest nightly?

kursatu, 2021-11-09:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/7501604/SuperSlicer.zip)

kursatu, 2021-11-09:
My latest build, with the latest sources pulled, still does not work.
If I pass the -datadir to an emoty directory next to the .exe it starts up.

kursatu, 2021-11-12:
I've made a build by pulling the latest sources and it works again.
Thank you.

closed 2021-11-12 by kursatu