supermerill/SuperSlicer #1836: Empty Layer Detected, no other slicer had issues

inspectionsbybob, 2021-11-09:
### Version
2.3.56
### Operating system type + version
Win10

### Behavior
When slicing this pencil holder I got this error.  I then Tried Cura with no errors or warnings.  Prusa slicer no errors or warnings.  So I let Cura-Mesh tools-FIX do its thing. and then saved it and re-imported it in to SS, where it sliced and is now printing (15-hour print).

Just Seemed Weird so I thought it was worth reporting.

![error2](https://user-images.githubusercontent.com/22258531/140849374-03554a6d-ae97-47ce-aa51-7ea92492cc0b.png
[treestump_printready_.zip](https://github.com/supermerill/SuperSlicer/files/7501409/treestump_printready_.zip)
)

supermerill, 2021-11-09:
what version of prusaslicer?

inspectionsbybob, 2021-11-09:
Version 2.3.1+win64

supermerill, 2021-11-09:
I can slice your project file without any error or warning on 2.3.56.9
are you sure it's the faulty one?

inspectionsbybob, 2021-11-09:
Yup, I just got it again.  I will re-re-load and send it again.

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2021-11-09:
Just confirmed it, and re-downloaded it into SS and sliced and cot the warnings.
[treestump_printready_error.zip](https://github.com/supermerill/SuperSlicer/files/7507327/treestump_printready_error.zip)
..

inspectionsbybob, 2021-11-09:
I will load the current release you just pushed out and see if it still does it...

Doing a backup at the moment

nirurin, 2022-09-05:
I am also having this same problem, but on multiple different models (including the calibration retraction model, if I add a raft). Doesn't seem to be related to any specific feature. I thought it might be an arachne bug, but it also happens with the classic engine. 
Never had the issue before I updated to the Aracne release, though I was a few updates behind before that so I don't know when the bug might have crept in.

nirurin, 2022-09-17:
Still happening for me.