supermerill/SuperSlicer #1837: wxWidgets Debug Alert when closing

ristein, 2021-11-09:
**Describe the bug**
Had two windows open with same print settings active, saved them in window two, pressed close on window 1, then this:
[image]
[image]

**To Reproduce**
as above

**>> Project File <<**
[Deckel.zip](https://github.com/supermerill/SuperSlicer/files/7503183/Deckel.zip)

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.57.4

supermerill, 2021-11-09:
I can't reproduce.
I think the problem may be in the "unsaved changes dialogs".
what are the fields you changed? and with what value?

ristein, 2021-11-09:
sorry, reproduced it now as follows:
It doesn't have to do with two windows but with importing a project:
1. open SuperSlicer
2. import project
[1766_Deckel_100_above_bridge.zip](https://github.com/supermerill/SuperSlicer/files/7504245/1766_Deckel_100_above_bridge.zip)
 3. change anything i.e. perimeters
 4. close

supermerill, 2021-11-09:
can't reproduce either.
can you zip and post here your configuration directory?
it can be found via help -> show conf folder
If you have physical printer with ip & password you don't want to share, you can export the config bundle instead, maybe it's enough (but maybe not).

ristein, 2021-11-09:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/7504420/SuperSlicer.zip)

I use microSDcard 🙈

supermerill, 2021-11-09:
Strange bug. I can only reproduce it with builds from github, not with builds from my computer.

ristein, 2021-11-09:
did u try zipping/unzipping?

supermerill, 2021-12-04:
seems like the mitigation works. can you try the latest nightly?

ristein, 2021-12-07:
seems to be working on v..57.7

closed 2021-12-22 by supermerill