supermerill/SuperSlicer #1842: Support Under Nothing on 45°| Model Support Gap

kpdcompany, 2021-11-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Load Arched Part and run 45 degree support under it
Turn model support to 0/90 parallel to wall to see large gap between model and support

**>> Project File <<**
NDA, Sorry, please see attached.  Could not replicate exactly with other model so this is what I am showing.

**Expected behavior**
Support goes well beyond model boundary on 45's and spacing between wall and support on 0/90 supports is too great when parallel.  Spacing is set to 0.3mm.

**Screenshots**

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Win10
 - SuperSlicer_2.3.57.5

supermerill, 2021-11-09:
> EDIT: I Would like an explanation on Bridge Flow. 

Bridges can't be squished as there is nothing under them, so it's a round extrusion
100% bridge flow => extrusion diameter = nozzle diameter
50% bridge flow => half the flow, so the extrusion diameter is equivalent to ~70% of the nozzle diameter (because it's squared)

overhangs can be printed with bridge flow, as they have nothing under them.
but if the overhangs can be over a half-periemeter. You can set the threshold at which the perimeter need the bridge flow. It's the number of mm from the end of the nearest perimeter under it. If yous et a %, you can see it as how much of the overhang needs to not be supported for it to get the bridge flow.

kpdcompany, 2021-11-09:
I think that makes sense, but why if I set it to 5% it is flowing more than 0% or 100%?  It seam no matter what the setting its just on or off, then this sets what it applies to.  I was not adjusting bridge flow as I think it will adjust my Overhangs?  Could we get a Bridge Infill / Bridge Perimeter Flow settings maybe?

supermerill, 2021-11-09:
i'll rework it after 2.4 merge, as prusa touched it.

kpdcompany, 2021-11-10:
something is up with bridge fill too, its not centered in the part or between perimeters, its also not overlapping like it should.  It was working last release I used
[image]
.

kpdcompany, 2021-11-10:
we have found another similar issue, please see supports where there is no model ever..
[image]
..

kpdcompany, 2021-11-10:
and another, this totally makes sense why all my support is really bad...the interface prints in mid air also with a gap between, the regular support, this is a HUGE issue.
[image]

supermerill, 2021-11-10:
I don't use support that much.
And i can't advise on a picture. If you give a project file maybe... but giving advice here and here isn't efficient.