supermerill/SuperSlicer #1847: Fan Speed management

5axes, 2021-11-10:
**Describe the bug**
I'm not sure if it's a bug, or something I don't understand. When I try to decrease the value of the Fan Speed (it's labeled Default speed but in fact it's a **min_fan_speed**   the fan speed used in some area of the part doesn't respect this value. 

[image]

**Expected behavior**
The fan speed should be for the wall part at the Default Fann Speed except for the first layer where we can have a lover speed

**Screenshots**
[image]

**Desktop :**
 - OS: , Windows
 - Version 2.3.57.5

**Additional context**
The fan speed calculation is different if we use as Filament settings based on  the ABS or PETG configuration.

**>> Project File <<**
Joined  project file

[Multi_Bridge.zip](https://github.com/supermerill/SuperSlicer/files/7514713/Multi_Bridge.zip)

5axes, 2021-11-10:
Ok I think I just understood the problem. The misunderstanding comes more from the logic of the parameters layout. I think it would be better to keep the order and the labels like in Prusa with Mini and Max speed.

**SuperSlicer**
[image]

**PrusaSlicer**
[image]

supermerill, 2021-11-10:
That 'maximum' is only useful for the "short layer time", and it isn't a maximum as it can be superseeded by the bridge, top and perimeter fan speed.
I moved it here because it was confusing for me to have a max value lost in the top panel that wasn't a max.
The important part is the explanation text below.

5axes, 2021-11-10:
Ok this behaviour will really deserve an explanation in the documentation  :)

closed 2021-11-10 by 5axes