supermerill/SuperSlicer #1850: 2.3.57 not processing old files anymore

stuwe911, 2021-11-11:
**Describe the bug**
I have this very large print that goes to the limit of my printer. the brim is at one point only 2 lines outside the physical bed. when slicing with 2.3.56 i got the error message "a toolpath outside the print area was detected" but he would still slice and let me export the gcode. when printing the printer runs into the end-switch and prints these 2 lines over each other. all the rest is fine and i need that brim.

Now opening the exact same project file in 2.3.57 when slicing i get a pop-up window "ERROR - Coordinate outside allowed range". once i click ok and then i get instead of "exporting gcode" a "invalid data" and game over. no code - no printing anymore! I actually had to go back to the old version to be able to print this file... which is not what you want to do when upgrading any software

I understand the problem the slicer has and it is good that he warns me. still i'm not a friend of being babysitted by the software and then denying the service. the warning is good but let me as a user take the risk and override it. it is not logical to me as of why this exact same file is printable in the old version but not in the improved one anymore... a step back in my eyes.

**To Reproduce**
Steps to reproduce the behavior:
1. goto 2.3.56 and take a big print
2. create a brim that is slightly outside your print bed
3. slice and export gcode - save the project
4. open 2.3.57 import the same project file
5. try to slice and watch..

**Expected behavior**
same as in the old version. launch a warning but slice and export - users risk
please revert that new function back to the old one giving the users maximum flexibility
sometimes 3D printing requires a little bit twitching and squeezing... please don't turn this into a baby-safe slicer

**Desktop (please complete the following information):**
- Windows 7
- Version 2.3.56.9 and 2.3.57.5

supermerill, 2021-11-11:
You increase your bed size (and moving the origin a bit in the inside).
I'll look to re-enable slicing even when a warning is here.

stuwe911, 2021-11-11:
I did that. went from real 35cm to virtual 40cm.. he does not print where he should. however slicing even if there is a error is much appreciated!!! I've tried many different tweaks but nothing worked. the only solution to get the gcode was going back to the previous version. thanks!