supermerill/SuperSlicer #1851: initial_layer_height - variable is not usable in start g-code

bahmrockk, 2021-11-11:
**Describe the bug**
Variable initial_layer_height throws an error when using it in Start G-Code
According to this documentation it should exist:
https://github.com/supermerill/SuperSlicer/wiki/Variable-list

Either the documentation is wrong, the discoverability or myself :-) 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Printer Settings"
2. Click on "Custom G-Code"
3. Enter anything that uses the variable "initial_layer_height" . In my case: 
`G1 Z[initial_layer_height]`

4. Error when Slicing:
`G-code export to /tmp/.5363.gcode failed due to invalid custom G-code sections:
start_gcode
Parsing error at line 14: Variable does not exist
G1 Z[initial_layer_height]
     ^
Please inspect the file /tmp/.5363.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.
`

**>> Project File <<**
See Attachments. Includes the complete g-code as well to give context on what I was trying.

**Expected behavior**
Either the documentation is adjusted, the variable  or myself :-) 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 2.3.56.9

neophyl, 2021-11-11:
I think it should be called G1 Z[first_layer_height] in which case the documentation is probably wrong.

You asked about discoverability, if you hover over settings the tooltip should tell you what the setting is called

supermerill, 2021-11-11:
yes, it's sla-only

bahmrockk, 2021-11-11:
first_layer_height is the percentage value of the first-layer deviation though (i.e. 75%), what I searched for was the 0.03 (instead of 0.02 for all other layers as saved in layer_height ).
I am not aware of doing arithmetic in the g-code though so I have no way of accessing the true first layer height in the machine g-code, correct?

Should I instead open a feature request? 

And thank you for the ultra fast reaction speed! <3

neophyl, 2021-11-11:
Given your settings isnt your first layer height 0.15 ? (0.2 *0.75 =0.15).  Why would you be searching for 0.03 ?

Any values output to gcode should use the results of the slicing math so would/should be G1 Z0.15.  However I notice if I slice your project with 2.3.57.0 its adding 
G92 E-0.2; Remove a potential blob
G1 Z75%
that looks like a defect to me.

bahmrockk, 2021-11-11:
I'm now even more confused - when I slice an object within the attached project, the first layer height move is:
; Filament gcode
M900 K0.8
;LAYER_CHANGE
;Z:0.3
;HEIGHT:0.3
G1 Z0.300000 F6000

I just looked for a variable to match the gcode height without realizing that I need really fancy math to get to 0.3 from 0.2 and 75%.
I guess this is something I'm not yet understanding from superslicer though and seems unrelated to the subject at hand.

Should I open a new ticket for the 75% or should I edit the original post? I'm not yet  familiar enough to know what a good practise is.

neophyl, 2021-11-12:
I'm not sure why its adding that G1 Z0.3 for the first layer.  I suspect its related to the defect with setting 75%.  

If you set an explicit value for first layer, for example 0.15 and slice then it puts G1 Z0.15 instead like it should.  

As Merill is already aware of this issue and as the first layer when set as a % is probably related to the incorrect height put in the gcode I don't think you need to change the title.  

If Merill feels it needs changing he can always do it to make it more specific to the actual defect.

supermerill, 2021-11-12:
if you set 50% for first_layer_height, then:
`[first_layer_height]` => `50%`
`{first_layer_height}` => `0.2`
`{if first_layer_height<0.4}Y{endif}` => `Y`

That applies to all % settings.

neophyl, 2021-11-12:
But it shouldn’t ever add a G1 Z75% into the actual output gcode should it ?  I don’t think printer firmware will be able to handle that.

supermerill, 2021-11-12:
write `G1 Z{first_layer_height}` and it will write the expected value.

neophyl, 2021-11-12:
When I tried it with the users project if you put 75% in as a value for the first layer in the print settings then it created the G1 Z75% in the gcode output as written above.

supermerill, 2021-11-12:
> When I tried it with the users project if you put 75% in as a value for the first layer in the print settings then it created the G1 Z75% in the gcode output 

Seems like my two previous comments were badly written, so i'll explain it differently.

When you have a field that can be a %, like `first_layer_height` (with for the exemple 50%), then if you are using the old, almost deprecated, kept only for compatibility braces like `[first_layer_height]`, then you have the old behavior of writing the value as-is, so it will write `50%`.
If you are using the now "standard" braces like `{first_layer_height}`, then you have the "new" behavior that will write a computed value, so `0.2`.

isn't it what you want? writing '0.3' istead of '75%'?

bahmrockk, 2021-11-15:
@supermerill you're correct, that is the intended behaviour! Thanks for the explanations and patience!

closed 2021-11-15 by bahmrockk