supermerill/SuperSlicer #1852: ETA not accurate in 2.3.57.4/5?

dragosmiron, 2021-11-11:
### Version
2.3.57.4/5

### Operating system type + version
MacOS Big Sur

### Behavior
After upgrading from 56.9 the ETA in the app for prints decreased, but I think it may be wrong.
The eta in 56.9 was more accurate. Now it is off (for a large print, in the example pasted) by ~4 hours

#### Project File (.3MF) where problem occurs
[part3-fixed.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7519313/part3-fixed.3mf.zip)

supermerill, 2021-11-11:
ETA for what? slicing or printing?
If you talk about the progress bar, i removed the old behavior that was a bit useless and replaced it by a new one where it show the progress of the current step.
But I agree, i should have keep the old one and split it.

dragosmiron, 2021-11-11:
i am talking about printing sorry.
For examole on the older app, (for a different stl) i got an estimate of 11 hours, then on the newest one, i got 9 hours, and then i thought "oh cool, this new app maybe has some impriovements so definitely i'm printing the 9 hour one!". But then it finished in 11-12 hours. I also know that printers also have their own weight into the inaccuracy, but i'm just referring from what I noticed between the 2 versions of the app on the same stl.

dragosmiron, 2021-11-13:
Probably is not so critical, but if I'd have put the socket to turn off automatically with a time delta error of a reasonable 1h, I'd have had a failed print

TWIISTED-STUDIOS, 2021-11-15:
I have noticed something similar I was slicing a file on my wanhao profile and the print says it would be 9 hours, I started the print and it only took 2-4 hours at most

supermerill, 2021-11-16:
I loaded your object. (please  save your project from superslicer, as i don't have your setting if you don't do that).
2.3.56.9 give me 7h28
2.3.57.4 give me 7h22 (beause it prints less solid infill by default, see solid_over_perimeters)

staffert, 2022-01-17:
I have also been getting actual print times in the 50-100% longer than expected ranges. It is just VERY hard to plan a long job when your slicer says 16h and it takes 28h!! even on a small 1h45 job it can take 3h+......something went wrong with the estimation subroutines.
I'm using the latest release 2.3.57.9

supermerill, 2022-01-17:
@staffert 
please double-check that the printers limits are exactly what's inside your firmware. I don't have a CR10SPRO, so I can only rely on user to provide me the info.