supermerill/SuperSlicer #1862: Show options menu not fully visible

vidi2000, 2021-11-12:
**Describe the bug**
The Show options menu in Preview is truncated (names not visible)
**To Reproduce**
Steps to reproduce the behavior:
1. Go to Preview
2. Click on Show Options menu

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
A clear and concise description of what you expected to happen.

[**Screenshots**]
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 2.3.57.5

**Additional context**
Add any other context about the problem here.

liftbag, 2021-11-16:
I cannot reproduce it with the latest nightly on macos Monterey

supermerill, 2021-11-16:
That may be only reproducible on a small screen.

vidi2000, 2021-11-16:
> That may be only reproducible on a small screen

 it’s not present in previous versions and in any Prusa Slicer release

thiagolocatelli, 2022-07-07:
I do have the same issue, with various versions of the Super Slicer. Currently using 2.4.58.2. MacOS Big Sur, resolution 1440x900.

snugwarms, 2024-01-27:
still present on 2.4.58.5, macOS 14.1.1. scaling settings are set to "looks like 1536x960" 

[image]