supermerill/SuperSlicer #1866: Weird Error Msg, don't understand

inspectionsbybob, 2021-11-13:
### Version
2.3.57
### Operating system type + version
Win10

Working With Calibration Objects
Temp Tower, 2-,2+ Steps 10

Get "ERROR:  First Layer height can't be greater than perimeter extrusion width"

First time getting that, don't understand it either...  Layer Height vs width of a perimeter Wall  

Got a temp tower started and then editing, went to generate another tower after some tweaks, and got this, but don't know what I tweaked to cause it...  Don't see a Perimeter wall width, only Perimeters  Vertical shells and Horizontal Shells.  What am I looking for?

[Temperature calibration error.zip](https://github.com/supermerill/SuperSlicer/files/7532588/Temperature.calibration.error.zip)

supermerill, 2021-11-13:
first layer -> 60%

inspectionsbybob, 2021-11-13:
I had it set for 75%. that's why I am confused....

Will try again when I am at that workstation.

supermerill, 2021-11-13:
that's what I see opening your project: 
[image]

inspectionsbybob, 2021-11-14:
Ok, this says 71% correct, not 75% if I am reading the right field, that is
still more than the 60% the error msg said, or the skirt (got an error on
that one as well on a different try) either one of them was set to
default/auto of "0" or %100 so I am confused...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-11-14:
Ok, I went and looked at all the help, the default for that field is %140.... or 1.4 width...

That is NOT clear from the Help menus and such.   I was trying to make the First layer 75% of the height of the Nozzle, and I guess it needs to be .30mm x 0.56mm... or 75% x 140% 

Got lost between the two different % affecting the first layer.

Feature Suggestion "Would you like me to FIX THIS... Press [FIX IT]"
on many of the error msgs that are single-minded/solution.

inspectionsbybob, 2021-11-14:
Ok, now I am REALLY confused...

I set the First Layer Extrusion Width to the **Default** of  140%
I set the First layer HEIGHT to the **Default** of 75%
And I get the error message!  
This is similar to the error msg originally, but from a different calibration object, same basic issue...

I saved all the profiles and exited and tried again clean... the same thing happens...  

Extrusion width of 140% of .40mm is 0.56mm 
First layer height of 75% of .40mm is 0.30mm
I SHOULD have 0.26mm to spare before I get the error msg.

What am I missing, or what other modifier to layer height/width is there that I don't see?

I have enclosed screenshots and the 3MF project file

[Screenshot (4).zip](https://github.com/supermerill/SuperSlicer/files/7534118/Screenshot.4.zip)
[Flow  error.zip](https://github.com/supermerill/SuperSlicer/files/7534127/Flow.error.zip)

neophyl, 2021-11-14:
Your external perimeter width is set to 0.0215mm .  Your nozzle width is 0.4mm  Can you not see that that is about 1/5th of the nozzle diameter ?  You cant print an extrusion that thin practically.
Did you mean to type 0.215mm ?  even that width is far less than a 0.4mm nozzle can reliably print at.

So your external perimeters are set to print at 0.0215mm width and your layer height of 0.2.  Do the math, you are printing a layer that is nearly 10x higher than your extrusion width.  It cant be done so the slicer is giving an error.  That error does NOT mention the first layer, so why are you assuming that its related to the first layer ?

inspectionsbybob, 2021-11-14:
Ok, got it....
Screen shots don't lie, but I could have SWORN I set all of those to default, double click, and all that...
That seems to have worked, but that is a weird value, not something I would have typed in, so something got changed when something else was edited, I will be watching that setting, along with others to make sure when I edit ONE field, other don't change UNEXECTEDLY...

Again, the error message doesn't help, and mentions setting that aren't what is needed... I was fighting layer height and firs layer extrusion width, Perimeter width wasn't even on my radar...

neophyl, 2021-11-14:
Look at the field to the right, the width and spacing combo field.  It has a zero in there, typing in there automatically changes the field its linked to, in this case the external perimeter field.  You have a zero in there.  Its very likely you accidentally made that field zero and so it changed the perimeter extrusion width.

inspectionsbybob, 2021-11-14:
Yup....
Just went through that section AGAIN... and there are a lot of "dependent" settings that are "0" or "110%"  that seem to change to weird values when other things are edited.  140% should STAY a 140% not show some new dependent fixed value.  

Found a specific example! 
Editing the Extrusion width section, first column, changed everything to the DEFAULT values. As I did that the the SECOND column changed to FIXED VALUES not the percentages or "0" defaults I had set them too.  Trying to change the SECOND column to defaults messes with the first column.  

I get that you can set the width -OR- the spacing, but if I set it to DEFAULT, leave the OTHER alone and set a default if that's where I set it... gets confusing when the "automatic"  default of a "0" or a percentage turns another field that was ALSO set to default/automatic/percentage into a fixed value.

neophyl, 2021-11-14:
At the moment the width and spacing columns are linked and dependent on each other.  They are supposed to be.  It wasn't that long ago that we didn't HAVE the spacing columns at all.  Notice that PS does not.  There are questions all the time about the spacing math of PS/SuSi/Slic3r.  
Those spacing columns were added to allow users who wanted to set the spacing of perimeters rather than the width of perimeters.  I know there's been at least one request to unlink them to make them independent which would give even more control.  Wrong values in there though would then be a major problem.

Also note that 0 may mean default in some settings but doesnt always.  Sometimes it just means 0.  For extrusion width with a 0.4 nozzle setting 0 is the same as putting in 0.45.  At least for default perimeter width.  Notice that when you set the extrusion widths to zero you are saying use the default, which while based on the nozzle diameter isn't the same for all of them as some use x1.125 of the diameter and other use different multipliers. 
Spacing default isnt 100% though, its based on your extrusion width and theres some complex math behind it (which I have trouble following).  Its designed such that the cross section of the extrusion when laid down means that theres no space between them.  You are dealing with oval cross sections.  Look at the example picture which can be expanded in that section ( below the width settings).
What doesnt help is that I think some of the tooltips are now out of date from the way it behaves now.  If it says default value of xx, then thats probably wrong now.  Most of those even in PS aren't accurate anymore as they have been there since the old old old days.

The thing you have to remember about Susi is that it opens up ALOT more of the settings than PS does.  I almost hesitate to say it but its better suited for those who have a need for more control, ie those who have more experience with settings.  It doesn't hold your hand or restrict you as much as PS does.  To use a Marvel line, with great power comes great responsibility.  Its the USERS responsibility to check their settings.  
If you don't understand how something works then ask.  Someone will usually be along to explain (if they can lol).  I've been using it for 3 years now and there still much I can learn still.  Especially as its constantly evolving and changing.

inspectionsbybob, 2021-11-14:
I get this/that... and that is why I want to set MOST things to AUTO so I
get at least a print.  May not be perfect, but it will let me slice...

EXAMPLE: So if the default on something is %110 percent of the nozzle, and
the default spacing is %10 of the nozzle, you should be able to set BOTH to
"0" or "Auto". Right now if I try to edit the first column so it is the
default of "O" the second column gets set to a FIXED value of .04mm, NOT
auto... If I try to put the second spacing field back to auto, the first
field gets reset to a fixed value, which may be a percentage, but it does
not "appear" to "Auto" anymore, and is definitely not the default of "0".

Auto needs to mean auto, leave it alone and whatever the math says at that
moment is the value, not the fixed value based on the edit to the default
in a different field...  maybe auto can be "(A0.04)" meaning the field is
set to auto and the current calculated value is 0.04...

Maybe the extended help can be "The field is currently set to Automatic (A)
and the calculated value is "x.xx". To -fix- the value to this calculated
value double click the field"

 Percentages are great for this... so long as they don't mess up something
else. If a value defaults to 140% and I edit it to be 150%, any thing
downstream that is calculated on the field should REMAIN calculated, now on
the 150% value, not become a fixed value... When that REALLY messes things
up is when you change the first Field BACK to 140%, as now all the
downstream values used to be calculated of the first field, now have fixed
values in them.

What I think happened to me is I edited the first field and put in a MM
value rather than a percentage, oops, re-edited it correctly. But the FIRST
edit changed all the DEPENDENT calculated field to something FIXED based on
the incorrect value.  When I corrected the first field, the second field
DIDN'T update as it now had a fixed value it it that is absurd based on the
first edit of the first field... whew...

Back to auto... Auto needs to STAY auto, ESPECIALLY if the field it is
related/dependent on changes.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

neophyl, 2021-11-14:
What values do you think they are based on ?
Sorry but I don't think you have read the tooltips properly.  For example, with perimeters When set to 0 the value used is the same as that in default extrusion width.  If as a % then its based on nozzle diameter (in PS if set as a % its based on LAYER Height just to make it way more confusing and one reason why profiles arent 100% compatible between the 2 pieces of software).  Its NOT based on Default Extrusion width.  SO theres not really anything downstream as you term it except for the spacing which is linked.  What a value is based on can CHANGE depending on the units selected.  Now you may think that's confusing, and maybe it is but that is the way it currently is.

I'd suggest you just ignore the spacing fields completely.  As an example set all the widths to 100% and then slice a simple cube.  Look in the preview and select the type of preview to width.  You should see only 1 or 2 width values in the key.  All the extrusions should be coloured for 0.4mm (except maybe the bridge infill, when I did it as my bridge infill is set to 95% it was 0.39mm).  Changing bridge infill flow to 100% made all extrusions 0.4 at that point.  

Basically use the preview types and change each setting individually and observe what is sliced.  You will learn more and get a better feel for what is happening.

inspectionsbybob, 2021-11-14:
I reset Everything and am trying again...

Calibration cubes come out GREAT, it is when objects get tall and Skinny
(like supports) when I have my biggest problems.

I do apologize for my questions.  I am fighting a cantankerous printer with
KNOWN extrusion problems (A brand you have never heard of, JGMaker,  Mine
is a pre-production KickStarter version) and trying to
learn/UNlearn several slicers.  So many times when I THINK I have got a
profile NAILED, I try to print something other than a calibration cube, and
it goes to spaghetti FAST...  I have been watching lots of TUNING videos on
how to get the best print if you have an "X" problem... and then trying to
find where in YOUR slicer is a similar thing as most of the videos are for
CURA 4.11 or 4.12.

The NEXT thing I am fighting is supports... supports that fall over -OR-
are too close to the Model and leave marks...   Again, I get a profile
close, and I try it with my torcher object... lots of Spaghetti this week...

Each slicer has things the others don't.  Cura  has the most "Extensions"
and calibration Parts, so I am reluctant to get too far from it.  Your
latest load added a lot more of the PS stuff in.  I hope the PS people see
your work and fold it into their stream as well as they have the money to
maintain the project, where you may slow down when you graduate?

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
***@***.***
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

neophyl, 2021-11-16:
First I'd just like to point out that its not my software.  I am just a user of it.  The Developer is SuperMerill.  I just try to help out where I can, mainly answering questions as that theoretically gives more time for them to develop the software.

Also don't apologize for having questions, need to ask questions to learn new stuff.

Spaghetti is all down to bed adhesion.  Problems with that are usually caused by improperly set first layer distance (ie adjusting your bed height properly to get a good first layer squish) and bed cleanliness to remove all surface oils etc.  Or applying a suitable adhesive if you have to.  Sorry I'm not familiar with your printers build surface.  I've now modified all my printers including my Creality ones to use prusa style magnetic sheets with pei surfaces and since doing so have very little trouble with parts ever coming loose.

Supports are one area where Slic3r/PS/SuSi have always lacked compared to some other slicers.  That's not to say they cant do supports.  The new paint on enforcers make setting them up much much easier but there are still issues with their actual generation.  
You might have better luck posting a project file with your model and asking people how they would orient and do the supports for it.  That way they could modify the 3mf and post an updated one.  Its something we do a lot of on the Prusa forum for members having trouble.  You can examine what they have done and thereby see different support techniques and tricks used.  Things like separation of the support in x/y and also in z for ease of removal.

inspectionsbybob, 2021-11-16:
I am going through pains with my printer...

The JGMaker Artist-D, kickstarter version has a questionable extruder and
hotend.  Inconsistent is a kind word.

Just went through re-verifying eSteps and now flow.  My eSteps were spot
on, my flow was high, might have been -one- of my blob sources...  I
lowered my flow and my walls got thicker?  So something else is going on
with my flow...

Anyway, as for supports... don't remember the slicer, but there was a
slicer that had a better tree style support that hugged the model better
than the version that seems to flow all over the bed... but that is a ways
off for SS.

I re-printed my torture model, but did it in shorter pieces, with a
dramatically lower flow and it did much better... not perfect but better...

Managed to crash, as in instant path to desktop, SS several times, and
found a few more "inconsistencies" that are more likely documentation
issues... Without knowing -all- of the links with settings, it is hard when
you edit one thing and lots of -other- fields change in -unexpected- ways.
Flow, wall thickness, layer height, and many other field are so
interdependent that it is frustrating trying to tweak a moving target...

Anyway... I will keep working with SS and PS and Cura.  I like what SS is
doing, -and- have concerns about the longevity of the project... what
happens when when this teams graduates or gets new harder day jobs? So I
will stay close to PS and Cura as commercial products...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2022-10-03 by inspectionsbybob