supermerill/SuperSlicer #1867: 1st layer problem only happens in Super slicer

RhepTheSpring, 2021-11-13:
on the 1st layer super slicer is picking up a cavity which starts on the second layer (0.2mm) from the face I have tried the same file in Prusa slicer no problem and S3d again no problem below Prusa slicer 1st layer
[image]
[image]

supermerill, 2021-11-13:
project file?

odziuba, 2021-11-14:
> NetApp Security WARNING: This is an external email. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hallo Joachim ,

Liste anbei , in dem file waren die Quotas für 3 vserver drin , filerts1 , natora1s und natora1k , ich war mir nicht sicher welchen wir schon hatten , deshalb sind hier alle drei drin 😊

Gruß
Oliver

[Image removed by sender. prusa slicer 1st layer]<https://user-images.githubusercontent.com/27011892/141657421-d7f6c414-1ea1-43b7-add3-85de2f2d61de.jpg>
[Image removed by sender. super slicer 1st layer]<https://user-images.githubusercontent.com/27011892/141657426-8568c335-779d-45a9-81a0-f470267d9031.jpg>

quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target dbtaskes              -disk-limit  50GB -soft-disk-limit      -  
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target enfst1s1              -disk-limit  10GB -soft-disk-limit      -
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target fiola_rep_tst         -disk-limit 200GB -soft-disk-limit      -
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-awa_ent_app     -disk-limit  15GB -soft-disk-limit   14GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-cfgo_std_test   -disk-limit  80GB -soft-disk-limit   75GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-cfgo_test       -disk-limit  45GB -soft-disk-limit   40GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-dasststs        -disk-limit  50GB -soft-disk-limit      -
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-esb_archiv_entw -disk-limit   5GB -soft-disk-limit 4.50GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-esb_archiv_test -disk-limit   5GB -soft-disk-limit 4.50GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-esb_archiv_wart -disk-limit   5GB -soft-disk-limit 4.50GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-fiola_all       -disk-limit 500GB -soft-disk-limit  450GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-jbdatsts        -disk-limit  10GB -soft-disk-limit      -
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-nessusts        -disk-limit  25GB -soft-disk-limit   20GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-talci1s1        -disk-limit  50GB -soft-disk-limit      -
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-talents1        -disk-limit 100GB -soft-disk-limit   80GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-talints1        -disk-limit 100GB -soft-disk-limit   90GB
quota policy rule create -vserver filerts1 -volume vol100_ufs_nfs -policy-name default -type tree -target qtree-talkm1s1        -disk-limit 100GB -soft-disk-limit   80GB

quota policy rule create -vserver filerts1 -volume vol1548_idsz200_nfs -policy-name default -type tree -target idsz200 -disk-limit 100GB -soft-disk-limit 90GB

quota policy rule create -vserver filerts1 -volume vol1653_idse200_nfs -policy-name default -type tree -target idse200 -disk-limit 50GB -soft-disk-limit 40GB

quota policy rule create -vserver filerts1 -volume vol1654_idst200_nfs -policy-name default -type tree -target idst200 -disk-limit 50GB -soft-disk-limit 40GB

quota policy rule create -vserver filerts1 -volume vol1655_idsw200_nfs -policy-name default -type tree -target idsw200 -disk-limit 50GB -soft-disk-limit -

quota policy rule create -vserver filerts1 -volume vol1662_idsw201_nfs -policy-name default -type tree -target idsw201 -disk-limit 50GB -soft-disk-limit -

quota policy rule create -vserver filerts1 -volume vol1668_cbat010_nfs -policy-name default -type tree -target bsptest -disk-limit 10GB -soft-disk-limit 8GB
quota policy rule create -vserver filerts1 -volume vol1668_cbat010_nfs -policy-name default -type tree -target cbat010 -disk-limit 60GB -soft-disk-limit 50GB

quota policy rule create -vserver filerts1 -volume vol1675_idsx200_nfs -policy-name default -type tree -target idsx200 -disk-limit 50GB -soft-disk-limit -

quota policy rule create -vserver filerts1 -volume vol1702_hdbn201s_nfs -policy-name default -type tree -target hdbn201s_sap    -disk-limit 50GB -soft-disk-limit -
quota policy rule create -vserver filerts1 -volume vol1702_hdbn201s_nfs -policy-name default -type tree -target hdbn201s_sapx10 -disk-limit 50GB -soft-disk-limit -
quota policy rule create -vserver filerts1 -volume vol1702_hdbn201s_nfs -policy-name default -type tree -target hdbn201s_sapx20 -disk-limit 50GB -soft-disk-limit -
quota policy rule create -vserver filerts1 -volume vol1702_hdbn201s_nfs -policy-name default -type tree -target hdbn201s_sapx50 -disk-limit 50GB -soft-disk-limit -
[... cut ...]
quota policy rule create -vserver natora1s -volume vol3722_orat340_nfs -policy-name default -type tree -target orat340_arc -disk-limit10GB -soft-disk-limit  -
quota policy rule create -vserver natora1s -volume vol3722_orat340_nfs -policy-name default -type tree -target orat340_rdo -disk-limit 8GB -soft-disk-limit  -

quota policy rule create -vserver natora1s -volume vol3726_orae350_nfs -policy-name default -type tree -target orae350_arc -disk-limit 20GB -soft-disk-limit  - 
quota policy rule create -vserver natora1s -volume vol3726_orae350_nfs -policy-name default -type tree -target orae350_rdo -disk-limit 10GB -soft-disk-limit  -

odziuba, 2021-11-14:
Sorry wrong mailaddy ☹

RhepTheSpring, 2021-11-14:
[Timex watch Top 4-0-clock.zip](https://github.com/supermerill/SuperSlicer/files/7533801/Timex.watch.Top.4-0-clock.zip)

neophyl, 2021-11-14:
Not sure why but its seems to be caused by the 1 perimeter on top surfaces option.  Turn that option off and it goes away and slices the bottom without those perimeters.
Its the first option I disable when I see anything weird.

RhepTheSpring, 2021-11-14:
neophyl,

Thank you,  I unticked and it went away, thank you still very new to Super Slicer. Again appreciated...

supermerill, 2021-11-16:
That is because the "inside" of the circle is a top surface (but also a bottom one). So the area is excluded for the second perimeter round.

Nyegaard, 2022-12-02:
@supermerill this issue is also active for "on overhangs" - is that a known bug or a sub-set of this same issue ? No need to open a new issue if it's the same thing :)