supermerill/SuperSlicer #1871: Calibration prints throwing errors on Linux

leetwanker, 2021-11-14:
### Version
_Version of SuperSlicer used goes here (help->about)_

2.3.57.5

### Operating system type + version
_What OS are you using, and state any version #s_

Linux Mint 19.3

### Behavior
* _Describe the problem_ 

When trying to run calibration prints on linux:

(superslicer:30024): GLib-GObject-CRITICAL **: 20:28:30.652: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[2021-11-13 20:28:31.332345] [0x00007f19ec7c9700] [error]   Post-processing script "C:\\Python39\\python.exe" "C:\\Manual_Install\\pp.py" on file /tmp/.SuperSlicer.upload.ea87-1cef-35b7-4c27 failed.
Error code: 127
Output:
/bin/bash: C:\Python39\python.exe: command not found

* _Steps needed to reproduce the problem_

I've tried the downloadable .AppImage and also compiled my own with the BuildLinux.sh -dsi script.

leetwanker, 2021-11-14:
Getting a similar error when trying to send a print to Octoprint: 
```

[2021-11-13 20:44:44.211867] [0x00007fbc063fc700] [error]   Post-processing script "C:\\Python39\\python.exe" "C:\\Manual_Install\\pp.py" on file /tmp/.SuperSlicer.upload.a6e1-94c9-0adb-7a7e failed.
Error code: 127
Output:
/bin/bash: C:\Python39\python.exe: command not found

```

csch0, 2021-11-15:
But do you have python installed? `C:\Python39\python.exe`

I think you used a profile from somebody else and a post-processing script (Printer Settings -> Output Options) is still set and not present on your system.

[image]