supermerill/SuperSlicer #1872: Bridge overlap and infill overlap bug and other setting inconsistencies  - please fix as soon as possible

thwerks, 2021-11-14:
"Bridge Overlap" setting does not change the overlap of the bridge lines/pattern, it seems to do nothing. On the other hand, "Infill/Perimeters Overlap" settings affects both bridge overlap and infill overlap at the same time but not in the same amount. This is extremely frustrating and has lead to many failed prints. Please fix this because i cant use SuperSlicer for ANYTHING that needs Bridging over infill. Bridges over sparse infill need MORE OVERLAP or the lines get detached and catch the nozzle. 

There are multiple problems:

1. Infill/Perimeters Overlap is not accurate in setting the infill overlap. %60 setting and the infill it's already %100 overlapped.
[image]

- setting Infill/Perimeters Overlap above %40 produces very strange effects. solid infill line spacing changes among other things

[image]

- For example, here %60 Infill/Perimeters Overlap gives about %25 overlap on bridge (which is pretty good) but its %100 overlap on infill as seen on earlier picture.

[image]

Now also at %60 the solid infills over bridged lines are way too far spaced, this leave a horrible surface.

[image]

- Now setting %25 Infill/Perimeters Overlap gives about %25 overlap on infill and nearly %0 for bridge overlap.

[image]

Infill looks ok and and solid infill has good spacing, so far so good,...

[image]

But now bridge doesnt have enough overlap, this is not good at all

[image]

2. Bridge Overlap does absolutely nothing and instead Infill/Perimeters Overlap affects both infill and bridge overlap, but in different amounts. 

3. changing bridge flow changes extrusion width and spacing but they are out of sync. setting Bridge Flow to %100 neither gives a %100 bridge line nor does it give %100 spacing, This is completely arbitrary and meaningless. %200 Bridge Flow give a %100 bridge line and %100 spacing. 

[image]

[image]

- In Prusa Slicer there is no direct setting for Bridge Overlap such as in SuperSlicer however the Infill/Perimeters Overlap does affect both Infill and Bridge overlap. Difference is in Prusa Slicer the overlap of infill and Bridge are affected equally, in SuperSlicer its unequal. 

Setting Infill/Perimeters to %60 in PrusaSlicer

[image]

Infill overlap looks good at about %60, its correct

[image]

Bridge overlap look pretty good to, about %35-40 it looks, at least its much better than SuperSlicer

[image]

**How it should be:**

1. If Infill/Perimeters overlap setting is going to adjust Bridge and Infill overlap, then it should be more in sync like it is in Prusa Slicer or they should be independently set such as in SuperSlicer but its not working 
2. Infill/Perimeters overlap should not have any effect on solid infill such as it currently does in SuperSlicer, this is a malfunction
3. Bridge flow set at %100 should be %100 of the nozzle width assigned for the extruder, this would be much more meaningful.
4. Bridge flow setting should sync extrusion width and spacing such as it is in PrusaSlicer or be independently set. Getting a random mix of extrusion width and line spacing such as it is in SuperSlicer is completely nonsensical.

Setting %50 Bridge Flow in Prusa Slicer actually gives correct  extrusion width and line spacing
[image]

[image]

In SuperSlicer %100 Bridge Flow actually gives %50 flow and %100 spacing. But, %200 gives %100 flow and %100 spacing. that doesnt make ANY SENSE

[image]

[image]

neophyl, 2021-11-14:
It would have been better to supply the requested project file when creating an issue but I would have to agree that something is most definitely up with overlap settings.  
I haven't looked at all the types like the OP has but the internal bridge infill is definitely not working the same.  I compared the latest 2.3.56 and 2.3.57 and 56 has the bridging lines next to each other and 57 has gaps.  I also noticed that the supporting dense layer which in 56 is spaced apart to give a sort of wide grid  is using the same spacing as the bridge infill itself in 57.  I also normally run my bridges with 110% overlap and that setting isn't changing anything.

thwerks, 2021-11-14:
Just add a cube by right clicking on the bed and issue are just right there. infill overlap and bridge overlap is bread and butter functionality for a slicer and its not even workable in Superslicer. I like the concept and initiative behind SuperSlicer but there is no way I could recommend it to someone. Anything that requires bridging over sparse infill will fail because bridge settings really have to be dialed in perfectly and there's no way of doing that with Superslicer. 

I'm testing with 2.3.57 but I remember in 2.3.56 that Bridge Overlap actually affected bridge flow and not the line spacing as I would consider more to be "overlap". I have never seen flow/extrusion expressed as "overlap" anywhere but in SuperSlicer and it was so confusing. Maybe the lead developers are not natively speaking English. Anyway mute point as in 2.3.57 Bridge Overlap does absolutely nothing.

closed 2021-12-15 by supermerill