supermerill/SuperSlicer #1874: Configuration modifiers/options selector for "Printer settings", "Filament settings" & "Print settings"

DzzD, 2021-11-14:
**Problème description :** 

If we need two printers even with only one prameter different it requiere to create two printers even if only one parameter is different (nozzle diameter, retraction DD,...)

The same is true for "Filament setting" & "Printing setting".
For exempe the same filament could be printed at 180° and 200° depending on situtation, or Printing setting may require a slightly change in some situation like having an option "Print faster".

**One possible solution :** 

The idea would be to enable for each of those items "Printer Setting", "Filamaent Setting", "Printing Setting", a new variant/option selector that will enable to change  some parameters. Users would be able to create as many option they wanted.

For example: 
 - For "Printer Setting" you would be able to create an option "*Direct Drive" that will overide de retration setting
 - For "Filament Setting" an option could be created like "Print hotter" that will set a different temperaure
- For "Print Setting" it could be print faster, this will overide all speed parameters 
etc....

This way user could create multiple options for the three differents tabs (printer, filament, printing) rather than duplicate profile.

How it may work :
 - a selector at the top right corner to select option (there will always be at least one option "default")
- a button save/new/erase next to the option selector

Not sure how hard it is to implement but it will definitly make Super Slicer a lot more powerfull for congfigurations and ahead of the other slicer.

supermerill, 2021-11-16:
Could be nice, but it's a humongous task at first glance.
The ui and all the config system have to be rewritten in a not trivial way.

DzzD, 2021-11-16:
I dont know yet how it is implemented as I did not deep into the code yet, but maybe there are some approach that could make it easier than it appear at first glance.

I saw that settings are stored raw in text file, xml would have been great help and more flexible but it is as it is, maybe by playing with filename it could do the job.

like for xx/printer/Geeetech A20M.ini we could have :
Geeetech A20M-1.ini

sprior, 2021-11-17:
@DzzD I'm trying to learn how to do this myself, but take a look at the built in Artillery profiles for the Sidewinder X1 and I think it's called the Genius.  The profiles have a base layer with all the common settings for both and then the profile for the specific printer overrides what it needs.  I don't know yet how to create such a structure, would be amazing if the GUI made that easier, but it is possible.

DzzD, 2021-11-17:
What you said is that It is already possible by modifiing the file but not with the GUI ??

sprior, 2021-11-17:
@DzzD Not only possible but that there is an example you can study, but I don't think you can do it with just the GUI.

DzzD, 2021-11-17:
After deeping into the code and with the help gived to me by Merill, I think I can do the GUI, just need some time but dont want to do this if someone else is already further than I am

supermerill, 2021-11-17:
vendor profile can inherit other profiles.
So in most vendor profiles, you have a base profile and each printer profile is the nearest profile with one or two modifications.
But each printer is presented as a different one in the interface. There is no GUI to handle inheritance.

DzzD, 2021-11-17:
Profile inheriting of others profiles is quiete similar but not the same, but the fact there are already a system of inheritance can be helpfull to implement those options, I will start with "Print settings" and one setting, if it works than we will be able to add any other settings.

The idea of the GUI is to put up-right corner (same height as profile selector) a select with save/erase button, the selector will always have first option equals to "Base" and set by default chen changing profile, other entries will be the user specific overide, like :
Options : "Base Parameters"
                "Slowest"
                "Fastest"
                "etc..."

DzzD, 2021-11-18:
that a start
[image]

DzzD, 2021-11-18:
About IDE, I am using Visual Studio Code and find it very convenient (I use it for other language & aplication too), but if someone know how to add a button "build&run" I would be very happy, for now I have two separated buttons, and as the compilation take from 30s to 1mm it is not very practical

DzzD, 2021-11-19:
I am having hard time with the understanding of part of the source code, especially those Preset concept that is used... It may take a bit more of time but I am working on this new feature and the work is progressing...

(Any help about special tips that I should be aware on Presets are welcome)

supermerill, 2021-11-19:
I think you have to create a new kind of preset, with all the tooling associated with it.