supermerill/SuperSlicer #1875: Variable for config dir (Bed Texture)

tome9111991, 2021-11-14:
Hi
I have a Linux and Windows maschine
I load config dir from my Seafile cloud.

But have issue with Bed Texture.
Have set on my Windows Pc path C:/....
Linux can't find the Path..

Here a Screen from my Linux Ubuntu Mate:
[image]

My idea
Copy the Bed texture in config dir on my Seafile cloud.

With Variable you can open the Config dir and choose the texture.

This cloud be fix my problem.

tome9111991, 2021-11-14:
Could be usefull for post-process script

tome9111991, 2022-07-18:
i have seen this.
https://github.com/supermerill/SuperSlicer/issues/2925

maybe possible to use with bed texture?

supermerill, 2022-07-18:
A bit more difficult, but okay.

supermerill, 2022-07-19:
It will save only the relative path if the file is in the exe, configuration or user directory/subdirectory tree.

tome9111991, 2022-07-20:
thanks works well.

i have copy bed texture in my config folder.
on linux i have set the path to bed texture.svg

windows will use path from config dir.

closed 2022-07-20 by tome9111991