supermerill/SuperSlicer #1876: 0.6 Nozzle doesn't print corners

Frasta37, 2021-11-14:
**Describe the bug**
[image]
[image]
I noticed that when I change the nozzle setup moving from 0.4 to 0.6 a print lost corners.

**Expected behavior**
With the 0.4 nozzle setup the shape was normal completing the corners. See pics 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Version 2.3.57
**Additional context**
Add any other context about the problem here.

neophyl, 2021-11-14:
Without your saved project file it’s impossible to be certain but just looking at the geometry those areas of the model are likely to be thinner than your extrusion width of the 0.6 nozzle profile. 
So they won’t be sliced. The best way to fix is to alter the model so that all walls are printable at the extrusion width you want to print at. 
If you can’t alter the model then you can try the detect thin walls setting. This will try to print details thinner than the extrusion width. If using this setting please check the preview in all areas carefully as it can introduce its own issues. 

An alternative method if absolute printed size is not important is to use a positive x/y compensation. This will grow the perimeters out by whatever you set. It’s quite useful for picking up too small details on sculptures but not good if doing mechanical parts that need to fit together.

Frasta37, 2021-11-14:
> Without your saved project file it’s impossible to be certain but just looking at the geometry those areas of the model are likely to be thinner than your extrusion width of the 0.6 nozzle profile. [...]

Ok thanks, I was thinking the same.. it's caused by the design. It works only with the extrusion width lower than 0.40.  
[draken-1.7.zip](https://github.com/supermerill/SuperSlicer/files/7533721/draken-1.7.zip)
this is the design.

supermerill, 2021-11-16:
this is the design.

You have to use `File` -> `save as` to create a good project file.
then zip it and drop it here.

maybe thin walls can help, but I doubt.