supermerill/SuperSlicer #1879: import file via url

CarlosRodriguess, 2021-11-14:
the goal would be to import files via URL links, to avoid cluttering up your hard drive with files that in the end we will print, the idea will be to copy the links on sites like thingiverse, youmagine, prusaprinter or other

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2021-11-16:
a download link, that end with .stl or other valid extension, and that can be downloaded by a simple wget?

CarlosRodriguess, 2021-11-16:
Would there not be a way to suggest when closing superslicer whether or not to save the file, as it does for settings and projects?

supermerill, 2021-11-16:
if you save the project, the object is inside anyway.

CarlosRodriguess, 2021-11-16:
Seulement si on enregistre, libre à chacun d'enregistrer ou non