supermerill/SuperSlicer #1881: Crash while generating "Extruder retract calibration"

SphaeroX, 2021-11-14:
**Describe the bug**
i try to generate a Extruder retract calibration with these settings:
Step: 0.2
Height: 15
Temp: 207
When i press "generate" SuperSlicer will crash

**To Reproduce**
Question: Where can i find an error log or something else from the crash?

**>> Project File <<**
[crash.zip](https://github.com/supermerill/SuperSlicer/files/7534517/crash.zip)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 2.3.57

neophyl, 2021-11-14:
You could start from the command line using superslicer_console.exe --loglevel=9

SphaeroX, 2021-11-14:
thank you
[log.zip](https://github.com/supermerill/SuperSlicer/files/7534752/log.zip)

SphaeroX, 2021-11-15:
today i tryed with an height of 10mm and it works fine....

SphaeroX, 2021-11-15:
works today also with 15mm, maybe the windows restart helped

closed 2021-11-15 by SphaeroX