supermerill/SuperSlicer #1883: Calibration generate something off the bed dimension

erictioh, 2021-11-15:
**Describe the bug**
The generate calibration object consider center of bed as Y=0, X=0, Z=0. this is not correct because it will cause error to the printing the object go beyond the bed limit.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Calibration'
2. Click on 'Calibration Cube'
3. Click on "Generate"

**Expected behavior**
The generated object should have consider my machine limit, bed dimension/etc.

**Screenshots**
See the object is in the middle of bed, why the xyz is near 0 ? it should be around 150 based on my bed's dimension of 300x300
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 11]

supermerill, 2021-11-16:
The object is created by the calibration at the 150,150 position. So the slicer don't need to move it, or not that much.

erictioh, 2021-11-16:
> The object is created by the calibration at the 150,150 (center) position. So the slicer don't need to move it, or not that much.

superslicer created it in the middle i have no problem on that. but when i try to move the cube all the way up to edge of Y, the position is showing 300+150 = 450 why ? Sending such gcode to my printer immediately trigger print error, and you still think this is normal ?

supermerill, 2021-11-16:
can't reproduce
[image]